littlefs-project/littlefs #2: Enhancement: List a directory content

guillaumerems, 2017-12-29:
I just implemented the littlefs and it would be great to have a function to list a directory with the following result:
- type (file, dir)
- name
- size

geky, 2018-01-03:
Hi @guillaumerems, glad to hear you were able to get littlefs working.

You can use the `lfs_dir*` functions to list a directory like this:

``` c
int lfs_ls(lfs_t *lfs, const char *path) {
    lfs_dir_t dir;
    int err = lfs_dir_open(lfs, &dir, path);
    if (err) {
        return err;
    }

    struct lfs_info info;
    while (true) {
        int res = lfs_dir_read(lfs, &dir, &info);
        if (res < 0) {
            return res;
        }

        if (res == 0) {
            break;
        }

        switch (info.type) {
            case LFS_TYPE_REG: printf("reg "); break;
            case LFS_TYPE_DIR: printf("dir "); break;
            default:           printf("?   "); break;
        }

        static const char *prefixes[] = {"", "K", "M", "G"};
        for (int i = sizeof(prefixes)/sizeof(prefixes[0])-1; i >= 0; i--) {
            if (info.size >= (1 << 10*i)-1) {
                printf("%*u%sB ", 4-(i != 0), info.size >> 10*i, prefixes[i]);
                break;
            }
        }

        printf("%s\n", info.name);
    }

    err = lfs_dir_close(lfs, &dir);
    if (err) {
        return err;
    }

    return 0;
}

lfs_ls(&lfs, "/");
```

However, there are many different ways to list a directory, so this function probably won't end up in this repo.

This repo's goal is to provide a simple low-level (posix-like) implementation of littlefs that contains all of the core functionality needed for a filesystem. User-friendly bash functions (such as ls, cp, recursive remove) are left up to higher-level APIs.

guillaumerems, 2018-01-03:
Thanks @geky it works perfectly
I will continue the tests

jeot, 2021-07-16:
Hello all,
I was looking for a simple arduino/c++ solution to use within the NodeMCU module.
I noticed the solution provided was a little complicated for me and maybe related to modifying the LittleFS library itself (which I'm not sure, I'm just guessing).
So I have come up with a simple function for listing all the files inside a directory (recursively). 
I hope it helps future noobies like myself! :D

```
void listAllFilesInDir(String dir_path)
{
	Dir dir = LittleFS.openDir(dir_path);
	while(dir.next()) {
		if (dir.isFile()) {
			// print file names
			Serial.print("File: ");
			Serial.println(dir_path + dir.fileName());
		}
		if (dir.isDirectory()) {
			// print directory names
			Serial.print("Dir: ");
			Serial.println(dir_path + dir.fileName() + "/");
			// recursive file listing inside new directory
			listAllFilesInDir(dir_path + dir.fileName() + "/");
		}
	}
}
```
Use like this:
```
if (LittleFS.begin()) {
	listAllFilesInDir("/");
}
```
The output should be something like this:
```
File: /asset-manifest.json
File: /favicon.ico        
File: /index.html
File: /manifest.json      
File: /robots.txt
Dir: /static/
Dir: /static/css/
File: /static/css/main.1635e87c.chunk.css
Dir: /static/js/
File: /static/js/2.2c405b20.chunk.js
File: /static/js/3.8cb6885c.chunk.js
File: /static/js/main.435818a8.chunk.js
File: /static/js/runtime-main.00e2c71a.js
```

BenBE, 2025-10-02:
The `lfs_dir_read` function internally already continues where it left of at the previous step. Thus a small wrapper like the following:

```c
/* callback returns 0 to continue, negative to stop and propagate an error */
typedef int (*lfs_dir_cb_t)(lfs_t *lfs, const struct lfs_info *info, void *cb_data);

/* iterate directory and call cb for each entry */
int lfs_dir_iterate(lfs_t *lfs, const char *path, lfs_dir_cb_t cb, void *cb_data) {
    lfs_dir_t dir;
    int err = lfs_dir_open(lfs, &dir, path);
    if (err) {
        return err;
    }

    struct lfs_info info;
    while (true) {
        int res = lfs_dir_read(lfs, &dir, &info);
        if (res < 0) {
            lfs_dir_close(lfs, &dir);
            return res;
        }
        if (res == 0) {
            break;
        }

        int cbr = cb(lfs, &info, cb_data);
        if (cbr < 0) {
            lfs_dir_close(lfs, &dir);
            return cbr;
        }
    }

    err = lfs_dir_close(lfs, &dir);
    if (err) {
        return err;
    }

    return 0;
}
```

will mostly already do what you want. That wrapper can be used like this:

```c
/* sample callback that prints type, human-friendly size, and name */
int print_entry_cb(lfs_t *lfs, const struct lfs_info *info, void *cb_data) {
    (void)lfs;

    static const char *prefixes[] = {"", "K", "M", "G"};

    switch (info->type) {
        case LFS_TYPE_REG: printf("reg "); break;
        case LFS_TYPE_DIR: printf("dir "); break;
        default:           printf("?   "); break;
    }

    for (int i = (int)(sizeof(prefixes)/sizeof(prefixes[0])) - 1; i >= 0; i--) {
        uint32_t shift = 10U * (uint32_t)i;
        uint32_t thresh = (shift >= 31) ? 0x80000000U : (1U << shift);
        if (i == 0 || info->size >= thresh) {
            unsigned val = (shift >= 31) ? info->size : (unsigned)(info->size >> shift);
            printf("%*u%sB ", 4 - (i != 0), val, prefixes[i]);
            break;
        }
    }

    printf("%s\n", info->name);
    return 0;
}
```

```c
/* call to iterate root and print entries */
int rc = lfs_dir_iterate(&lfs, "/", print_entry_cb, NULL);
if (rc) {
    /* handle error */
}
```

The bigger issue most likely is with the internal caching and the requirement that some parts of the directory need to be read multiple times (due to CoW and figuring out the most recent state of things).

@geky I think this thin wrapper might be worth adding in the library itself. Or with a separate recursive variant.