littlefs-project/littlefs #5: Current status

SadE54, 2018-01-05:
Hi,

Whats is the current status of LittleFS ? Could it be already used in production environment ?

Regards,

Yann

geky, 2018-01-11:
Hi Yann, sorry for late response I've been offline,

What do you look for before considering a project ready for production?

I would say yes. LittleFS is beginning to be used by several projects, and there aren't any major changes planned. Though there is still a decent rate of issues being reported as LittleFS begins seeing use.

guillaumerems, 2018-01-19:
Hi Yann,
Whatever the project you use from github, you must perform a lot of tests on your platform to be sure everything is ok with your project.
I started to implement littlefs on my product few weeks ago and I can say that it does the job properly, it is a simple FileSystem which cover 99% of the functionalities required for an embedded system.
There are lot of improvement to be done but more simpler the system is, more robust it is.