littlefs-project/littlefs #7: file truncation behaviour (LFS_O_TRUNC)

talon75, 2018-01-08:
Hi and thanks for the great work!
I believe that LFS_O_TRUNC should work differently: if a file gets opened with LFS_O_TRUNC set and closed without writing anything, no truncation will occur. Is this the intended behaviour?

I've fixed it, quick and dirty, by forcing LFS_F_WRITING on open:

```C
if (flags & LFS_O_TRUNC) {
    file->head = 0xffffffff;
    file->size = 0;
    file->flags |= LFS_F_WRITING;
}
```

Thank you,
Alberto

geky, 2018-01-11:
I think you're right. I also like your solution, although that should be LFS_F_DIRTY or else littlefs may end up writing out garbage.

Also a minor improvement is to only set LFS_F_DIRTY if the file is not already length zero (such as when the file is new).

I can go ahead and push up a patch.

geky, 2018-01-11:
Here we go: https://github.com/geky/littlefs/commit/472ccc420343b7fe3463445dafad60f130cf027e

closed 2018-01-21 by geky