littlefs-project/littlefs #8: Support truncate function

roceh, 2018-01-11:
If possible support for truncate would be useful, e.g.

`int truncate(const char *path, off_t length);`

or using the file handle like ftruncate

geky, 2018-01-21:
Here we go:
https://github.com/geky/littlefs/commit/d88f0ac02ff93b549418d38d464b5591fce4f688

Let me know if you run into any issues.

s-macke, 2018-01-21:
Is it possible to update littlefs-fuse as well?

geky, 2018-01-23:
Sure thing, here you go: https://github.com/geky/littlefs-fuse/commit/ed8713cc0f72bca9e0327553aaea135b84d7fda5

Instead of using git submodules, littlefs-fuse is using git subtrees. If you wanted to switch to the most up to date branch, you could merge in littlefs's master yourself:
``` bash
$ cd littlefs-fuse
$ git remote add -f littlefs git@github.com:geky/littlefs
$ git subtree merge --prefix=littlefs littlefs/master --squash
```

The reason for not using submodules is that submodules adds an extra user step that isn't intuitive, whereas with subtrees all the user needs to do is clone and they're ready to go.

Either way, adding the truncate function required some modifications to the fuse wrapper.

closed 2018-01-29 by geky