littlefs-project/littlefs #9: Wrong pos cursor in append mode

guillaumerems, 2018-01-17:
Hi,

When opening a file with append flags, the pos cursor at beginning is 0 but should be the size of the file

I propose to add
```
if (flags & LFS_O_APPEND && file->pos < file->size) {
        file->pos = file->size;
    }
```

geky, 2018-01-18:
That's a good suggestion, but the current behaviour is standard (although not the most intuitive).

After append you will most like do one of two things:
1. write - in which case the pos will be moved to the end automatically
1. read - which would always return 0 bytes if pos was at the end of the file

If you want to have the pos at the end of the file you can put it there manually with seek:
``` c
lfs_file_open(&lfs, &file, "file.txt", LFS_O_RDWR | LFS_O_APPEND);
lfs_file_seek(&lfs, &file, 0, SEEK_END);
```

---

You can see the standard behaviour with a small program on your PC:
``` c
#include <fcntl.h>
#include <unistd.h>
#include <stdio.h>

int main() {
    int f = open("test.txt", O_WRONLY | O_CREAT);
    write(f, "aaaaaaaaaa", 10);
    close(f);
    printf("created test.txt\n");

    f = open("test.txt", O_RDWR | O_APPEND);
    int off = lseek(f, 0, SEEK_CUR);
    printf("pos after open %d\n", off);

    write(f, "aaaaaaaaaa", 10);
    off = lseek(f, 0, SEEK_CUR);
    printf("pos after write %d\n", off);
    close(f);
}
```

output:
```
created test.txt
pos after open 0
pos after write 20
```

more info on the behaviour of the O_APPEND flag:
http://pubs.opengroup.org/onlinepubs/9699919799/functions/open.html

guillaumerems, 2018-01-18:
Ok then we keep it as the default behavior