littlefs-project/littlefs #13: Support big-endian host

gmouchard, 2018-01-23:
Hi,

I'm currently running littlefs on a simulated PowerPC e200 VLE that is a big-endian processor.
I suppose that natural endian of littlefs file system is little-endian since most today ARM and x86 processors are little-endian.
Can you please include support for big-endian host machines?
Please find attached a patch for lfs.c (against commit commit d88f0ac02ff93b549418d38d464b5591fce4f688) to address the problem of endian.

Regards,
Gilles Mouchard

[lfs_c.patch.gz](https://github.com/geky/littlefs/files/1655905/lfs_c.patch.gz)

geky, 2018-02-04:
Hi @gmouchard,

I wondered when this would come. I didn't expect someone to put littlefs on a big-endian platform so soon!

Thanks for the pr and the patch. I changed it a bit to better fit in with the other intrinsics in lfs_util.h. If you get a chance, does this branch works for you?
https://github.com/geky/littlefs/compare/big-endian

One thing I'm not sure about is the macro used to check for the endianness. The `BYTE_ORDER` and `BIG_ENDIAN` defines didn't work for me (using gcc-x86_64-linux-gnu -std=c99). Reading up on gcc suggests `__BYTE_ORDER__` and `__ORDER_BIG_ENDIAN__`. Do you think these work or do we need more checks?

---

Some numbers:
```
geky@s:littlefs$ powerpc-linux-gnu-size lfs.powerpc.{1,2,3}.o
   text    data     bss     dec     hex filename
  16767       0       0   16767    417f lfs.powerpc.1.o  (commit d88f0ac)
  17347       0       0   17347    43c3 lfs.powerpc.2.o  (commit dbce536)
  17235       0       0   17235    4353 lfs.powerpc.3.o  (commit 67b4392)

geky@s:littlefs$ size lfs.x86.{1,2,3}.o
   text    data     bss     dec     hex filename
  17419       0       0   17419    440b lfs.x86.1.o  (commit d88f0ac)
  17577       0       0   17577    44a9 lfs.x86.2.o  (commit dbce536)
  17419       0       0   17419    440b lfs.x86.3.o  (commit 67b4392)
  
geky@s:littlefs$ arm-none-eabi-size lfs.arm.{1,2,3}.o
   text    data     bss     dec     hex filename
   9809       0       0    9809    2651 lfs.arm.1.o  (commit d88f0ac)
   9905       0       0    9905    26b1 lfs.arm.2.o  (commit dbce536)
   9813       0       0    9813    2655 lfs.arm.3.o  (commit 67b4392)

Sidenote: arm-none-eabi-gcc seems to chose slightly different code
for lfs_init.o unrelated to the endian changes, not sure why that is.
```

gmouchard, 2018-02-05:
Hi Christopher,

> Hi @gmouchard <https://github.com/gmouchard>,
>
> I wondered when this would come. I didn't expect someone to put
> littlefs on a big-endian platform so soon!
>
> Thanks for the pr and the patch. I changed [...]
I saw that you removed the source code in my patch that did a copy (in a 
local variable with a name prefixed by _) before swapping.
If you swap in place, be aware that the variable content is broken, and 
can no longer be used, until another swap.
For instance, in lfs_dir_update, I would write the following:

static int lfs_dir_update(lfs_t *lfs, lfs_dir_t *dir,
          const lfs_entry_t *entry, const void *data) {
lfs_entry_t _entry;
     _entry.d = entry->d;
     lfs_entry_eswap(&_entry.d);
     return lfs_dir_commit(lfs, dir, (struct lfs_region[]){
             {entry->off, sizeof(entry->d), &_entry.d, sizeof(entry->d)},
             {entry->off+sizeof(entry->d), entry->d.nlen, data, 
entry->d.nlen}
          }, data ? 2 : 1);
}

On a little-endian host, I think that an optimizing compiler will 
silently remove the copy, with no performance impact.

You may want to check your code on a virtual big-endian host using 
qemu-ppc and a crosstool-ng tool chain for PowerPC.
> One thing I'm not sure about is the macro used to check for the
> endianness. The |BYTE_ORDER| and |BIG_ENDIAN| defines didn't work for
> me (using gcc-x86_64-linux-gnu -std=c99). Reading up on gcc [...]
These checks are probably not sufficient for a full portability of the 
source code accross all platforms. The problem is that these macros 
(BYTE_ORDER, BIG_ENDIAN, LITTLE_ENDIAN) are provided by system headers. 
On most platforms it's sys/param.h.
Last week I realized that |BYTE_ORDER| and |BIG_ENDIAN |was not enough 
and that they break lfs on x86_64-linux-gnu (my PC was detected as a 
big-endian host) with fuselfs, so I've slightly modified my macros to 
repair. I would rewrite lfs_eswap function as the following:

static inline uint32_t lfs_eswap(uint32_t a) {
#if (defined(BYTE_ORDER) && defined(BIG_ENDIAN) && (BYTE_ORDER == 
BIG_ENDIAN)) \
  || (defined(__BYTE_ORDER) && defined(__BIG_ENDIAN) && (__BYTE_ORDER == 
__BIG_ENDIAN)) \
  || (defined(__BYTE_ORDER__) && defined(__BIG_ENDIAN__) && 
(__BYTE_ORDER__ == __BIG_ENDIAN__))
#if defined(__GNUC__) && defined(__GNUC_MINOR__) && (__GNUC__ >= 4) && 
(__GNUC_MINOR__ >= 3)
     return __builtin_bswap32(a);
#else
     return ((a & 0xff000000) >> 24) | ((a & 0x00ff0000) >> 8) | ((a & 
0x0000ff00) << 8) | ((a & 0x000000ff) << 24);
#endif
#else
     return a;
#endif
}

Regards,
Gilles Mouchard

geky, 2018-02-08:
Hi Gilles,

Sorry for the late replies, I'm currently juggling a few things but do want to get big-endian support in.

I did try your copying approach (commit dbce536) and tinkered around with the swapping in a few different places, but somewhat surprisingly swapping in place seemed to perform better in code size across all the compilers.

Also note in the patch I proposed (commit 67b4392) all endian swaps have matching swaps to restore the original content. Since littlefs is single-threaded, intermediate states shouldn't be an issue:
https://github.com/geky/littlefs/blob/67b439257b9700c5c3b61a3feb091010705ba531/lfs.c#L619-L628

The numbers I posted above were comparisons of code size between the copy and double swapping approach compared to master. To be fair, they are all using GCC, I'm not sure if there's a better compiler for PowerPC. Here's the code sizes:
```
powerpc-linux-gnu-gcc:
  16767  lfs.powerpc.1.o  (commit d88f0ac) <-- master (no BE support)
  17347  lfs.powerpc.2.o  (commit dbce536) <-- copying before swap
  17235  lfs.powerpc.3.o  (commit 67b4392) <-- double swaps

x86_64-linux-gnu-gcc:
  17419  lfs.x86_64.1.o  (commit d88f0ac) <-- master (no BE support)
  17577  lfs.x86_64.2.o  (commit dbce536) <-- copying before swap
  17419  lfs.x86_64.3.o  (commit 67b4392) <-- double swaps
  
arm-none-eabi-gcc:
   9809  lfs.arm.1.o  (commit d88f0ac) <-- master (no BE support)
   9905  lfs.arm.2.o  (commit dbce536) <-- copying before swap
   9813  lfs.arm.3.o  (commit 67b4392) <-- double swaps
```

My goal was to make sure the LE code size remained the same, but surprisingly swapping twice has better code size even on PowerPC (at least with powerpc-linux-gnu-gcc). I'm guessing this is just because byteswapping is only one instruction vs however many instructions are needed to push a whole structure on the stack, and maybe the poor optimization is due to pointer aliasing rules? I'm not sure but I couldn't get the compile to make the code any smaller with copying.

---

I'll check out qemu-ppc when I can. Maybe it would be a good idea to get that into Travis CI as well.

> The problem is that these macros (BYTE_ORDER, BIG_ENDIAN, LITTLE_ENDIAN) are provided by system headers. On most platforms it's sys/param.h.

That may be a problem. littlefs is targeting microcontrollers, where system headers are notoriously hard to come by. If we're going to include sys/param.h, I need it to be conditionally included based on some define that indicates sys/param.h is present. I even have a branch in progress that conditionally removes some of the "normal" system headers such as "stdlib.h" and "stdio.h":
https://github.com/geky/littlefs/compare/configurables

In the worst case, the manual bit-shifting at least works as a default on all platforms. (though unfortunately that too didn't get optimized out as I had hoped).

geky, 2018-02-22:
I went ahead and merged this into version 1.3 of littlefs. I also changed the `lfs_eswap` function to `lfs_fromle32` and `lfs_tole32` to make it a bit more clear what conversions are going on:
https://github.com/geky/littlefs/commit/a0a55fb9e5942b51331947b855756abcac42a3c5

I was able to use qemu-ppc + gcc-powerpc-linux-gnu to test and make sure everything was working, and added powerpc into the CI. So the powerpc build should be tested almost as well as the x86 build.
https://travis-ci.org/geky/littlefs/builds/344127503

I'm going to go ahead and close this for now, but feel free to reopen or create additional issues if this needs work. And let me know if the big-endian support works for you.

closed 2018-02-22 by geky