littlefs-project/littlefs #15: make test race + warning

iamscottmoyers, 2018-01-26:
I ran in to 2 issues when running `make test`.

The first issue is that the test.c file is generated, compiled and run so fast that the file timestamp doesn't change between the first and second test. So make doesn't rebuild and the second test is run with the first executable, and I get this:
```make test
./tests/test_format.sh
=== Formatting tests ===
--- Basic formatting ---
lfs_format: 0
--- Invalid superblocks ---
lfs debug: Bad block at 1
lfs warn: Superblock 1 has become unwritable
lfs debug: Bad block at 0
lfs warn: Superblock 0 has become unwritable
lfs_format: -52
test.c:103: assert lfs_format failed with -52, expected 0
```

The second issue is that my compiler gives warnings on statements like `res = res;`, these are turned in to errors by `-Werror` and some tests fail with:
```
--- Exhaustion test ---
test.c:120:13: error: explicitly assigning value of variable of type 'lfs_ssize_t' (aka 'int') to itself [-Werror,-Wself-assign]
        res = res;
        ~~~ ^ ~~~
```

After working around those issues all tests pass! :)

theotherjimmy, 2018-01-26:
> The first issue is that the test.c file is generated, compiled and run so fast that the file timestamp doesn't change between the first and second test. 

What a problem!

geky, 2018-01-29:
Haha, that's entertaining. Does this patch fix the generates-to-fast issue?
https://github.com/geky/littlefs/compare/fix-test-no-build

geky, 2018-01-29:
Interestingly, the self-assign warning was legitimate. Some of the tests were creating a variable that conflicted with the test generators variables.
patch: https://github.com/geky/littlefs/commit/3dadc4ff8c56325c88cab5a9ae640dc511dc1b9a

I'm going to move the -Werror to CI only. It's not useful for users to not be able to run the tests based on compiler difference.

Let me know if this branch works for you:
https://github.com/geky/littlefs/tree/fix-test-no-build

iamscottmoyers, 2018-01-29:
I added a comment on patch 09951d7. lfs also needs to be added to the .PHONY dependency list or it only rebuilds test.o and not the executable.

I tried using make -B which forces everything to rebuild, that works but the test run takes nearly 3 times longer.

Another option is to delete the test.o and lfs files in the compile function before running make. That seems to be as fast as the .PHONY solution.

It's a weird problem to have, I've never seen it before, kinda makes a case for using a build system that builds based on hashes rather than timestamps.

geky, 2018-01-29:
Ah, that's a good point, deleting those files is simpler. Does this work for you?
https://github.com/geky/littlefs/commit/4ddb6fdb416c87f51a4a8505179b58f235202485

Thanks for trying the patch, my machine isn't fast enough to cause the same problem.

It does make is weird to reason about what gets built when timestamps fail, so point for hash based build system. But I've yet to see a build system that's as simple as makefiles.

iamscottmoyers, 2018-01-29:
Need to check the files exist first, or catch the `OSError` exception. If I run `make clean; make test` it throws an exception on the first test because the files aren't there.

geky, 2018-01-30:
Ah, embarassing. Well good catch. Does it work for you now?

iamscottmoyers, 2018-01-30:
Yep, that branch works for me now. Thanks!

aldot, 2018-01-30:
As an alternative to the fix-test-no-build branch let me suggest

https://github.com/aldot/littlefs/commit/0a2db6bf1b5db49882299bb7151585e3fbf87282
---8<---
Instead of always generating a hardcoded 'test.c' file, create different
files per test. I.e. for test_thing.sh generate test_thing.c

Disambiguate test blocks directory to allow to run the test in parallel
(think make -j test QUIET=1 ) while at it. This cuts the testing-time in
half for me ( formerly ~ 30 secs sequentially, now ~ 15 secs parallel)
---8<---

cheers,

geky, 2018-01-30:
Merged the branch fix-test-no-build into master for now. The parallel testing looks interesting, and we can still add it if we want.

closed 2018-02-01 by iamscottmoyers