littlefs-project/littlefs #20: Linking failed

Sim4n6, 2018-01-30:
```
gcc -Os -I. -std=c99 -Wall -pedantic lfs_util.o lfs.o emubd/lfs_emubd.o  -o lfs
/usr/lib/gcc/x86_64-linux-gnu/6/../../../x86_64-linux-gnu/Scrt1.o: In function `_start':
(.text+0x20): undefined reference to `main'
collect2: error: ld returned 1 exit status
Makefile:48: recipe for target 'lfs' failed
make: *** [lfs] Error 1
```
I got this kind of strange error

Sim4n6, 2018-01-30:
never mind , I had to use make all or make test :+1:

Sim4n6, 2018-01-30:
to do a linking it needs a main function ...

aldot, 2018-01-30:
yea, that's an unfortunate default target. We should build the OBJ per default (I've done that in https://github.com/aldot/littlefs ) or build size per default or the like.

Sim4n6, 2018-01-30:
or make a main file that do nothing

geky, 2018-02-04:
`make test` will leave a test.c file in the folder if a test fails, so you can then jump into it with `make DEBUG=1` and `gdb ./lfs`. That's why the current default target tries to build an executable.

You can compile a library explicitly: `make lfs.a`

Do you know if there's a way to get make to build an executable if there's a main symbol, and build a library if there's not?

Sim4n6, 2018-02-04:
I will work on that

Sim4n6, 2018-02-05:
I found a solution, I made an all new Makefile for the source files, and a seperate makefile for tests.
I added support for  default making for lfs.a if no test.c exists, otherwise make the lfs.
the code is available on my repo https://github.com/Sim4n6/littlefs I will pull request when the regression errors are fixed.

geky, 2018-07-06:
Hi @Sim4n6, I added your Makefile fix to this guy: https://github.com/ARMmbed/littlefs/pull/70

Let me know if you see anything wrong with it.

closed 2018-07-10 by geky