littlefs-project/littlefs #30: "port" headers

sn00pster, 2018-02-20:
I see that in the past day or so you've moved quite a few functions out into lfs_util.h, although this is a good move, that file should only contain defaults and should include say another file called lfs_config.h which the user can use to define functions, if they're not defined in the lfs_config file then they're defaulted to whatever is the default state by the lfs_util.h file.

lfs_config.h never appears in the repository, a version say "lfs_config_default.h" could be provided to show how to override functions.

This means we can safely pull changes without having to patch in our changes from our previous lfs_util.h file and also any new additions will be defaulted to their state as normal.

FreeRTOS and mbed both use this kind of system.

Great project btw, already building the FS into our next project, have it up and running on our eMMC module.

geky, 2018-02-22:
Hi @sn00pster, thanks for raising and issue, sorry for the delayed response.

I've had good success with using git subtrees to manage dependencies, subtree merging takes care of local modifications during updates surprisingly well. But point taken for projects where some sort of merge strategy isn't an option.

What are your thoughts on this option:
https://github.com/geky/littlefs/compare/config-file

This would allow you to provide an lfs_config.h as long as you define LFS_CONFIG in your build system (-DLFS_CONFIG=lfs_config.h for example). In which case lfs_util.h (effectively lfs_config_default.h) will not be used. I also noted a suggestion of copying lfs_util.h -> lfs_config.h as a starting point.

Thoughts?

Also exciting littlefs is working for you on eMMC. Glad to hear the project is being useful.

sn00pster, 2018-02-22:
That would work too, I'll take a look in the morning when I'm back at work!  Thanks for responding.

Yesterday and today I wrote abstraction around the eMMC device and some associated classes and also my filing system abstraction, I now have our eMMC "partitioned" into 3 sections, one with a small fatfs section for USB MSC in boot loader, one small lfs "system" partition and one big partition which is general storage.

geky, 2018-03-17:
I think this was taken care of in 1476181bd17b4d9998b618f115fe64f3b86af0a5, feel free to reopen if I'm wrong

closed 2018-03-17 by geky