littlefs-project/littlefs #32: C++ include

sn00pster, 2018-02-22:
Can we have a:

#ifdef __cplusplus
extern "C" {
#endif

.....

#ifdef __cplusplus
}
#endif

in the include file so it can be cleanly included by a cpp file?

geky, 2018-02-22:
Why not specify `extern "C"` in the C++ file?
https://github.com/ARMmbed/mbed-littlefs/blob/33a2ca519a5b2c3b97db15852721081340a2fdb1/LittleFileSystem.h#L22-L24

Seems a bit strange that a C library needs to support C++ and not the other way around.

sn00pster, 2018-02-22:
That's what I do, I personally just find it neater using an ifdef and extern "C" combo inside header files as it means you can just include it blindly from C or C++ without having to think.

Guess it's personal preference.

sn00pster, 2018-02-22:
On a complete tangent, I was looking at the little fs class in mbed and I couldn't for the life of me see where it allocated space for the file structure, it just uses a void * and doesn't seem to allocate it anywhere.  Don't know if that was your code and I didn't dig that deep so I probably missed the important bit.

geky, 2018-02-22:
Ah, it allocates the file structure in `file_open` using C++'s new operator, which makes it a bit harder to find:
https://github.com/ARMmbed/mbed-os/blob/master/features/filesystem/littlefs/LittleFileSystem.cpp#L372

One of the trickier parts of file-memory-management is that the file structure must be deleted in `file_close`, even if an error occurs:
https://github.com/ARMmbed/mbed-os/blob/master/features/filesystem/littlefs/LittleFileSystem.cpp#L394

geky, 2018-03-17:
I'm going to go ahead and close this, feel free to reopen if you want

PR #74 (2018-07-13): Add C++ guards to public headers
Even though it is perfectly possible to surround `#include "lfs.h"` with `extern "C" { ... }` directly in the C++ files, the common practice is to have such C++ guards directly in headers, even if this is a C library. The maintenance cost is basically zero (just remember to add includes "outside" the guards and code "inside"), while the gains are important (at least in my opinion):

1. Generally all C libraries do that, so it is a common practice, expected by people developing in C++. This is also true for C system headers like `stddef.h` (see `_BEGIN_STD_C` an [... cut]
commit 577d777c (2018-07-13): Add C++ guards to public headers

Fixes #53
Fixes #32
closed 2018-03-17 by geky