littlefs-project/littlefs #33: Assertion when creating too much files in the same directory

schouleu, 2018-02-26:
lfs_file_sync: Assertion `entry.d.type == LFS_TYPE_REG'  failed

This happens when we create a directory and then a lot of files in the root directory.

The issue is reproduced by the attached test.
[test_multiple_files.sh.txt](https://github.com/geky/littlefs/files/1759395/test_multiple_files.sh.txt)
[test_multiple_files.log](https://github.com/geky/littlefs/files/1759394/test_multiple_files.log)

geky, 2018-02-26:
Good catch and thanks for creating a great issue. The test case helped tremendously.

It looks like it was an issue with the directory chain appending not updating the internal directory structure. This issue doesn't show itself in mkdir, since it doesn't use the directory structure afterwards. Unfortunately, the existing tests against extending the directory chain were all made using mkdir.

Does this patch look good to you?
https://github.com/geky/littlefs/compare/fix-multi-block-append-files

schouleu, 2018-02-27:
It's great! Thanks for the fix

closed 2018-03-03 by geky