littlefs-project/littlefs #35: Purpose of lfs_config.sync

jrast, 2018-03-07:
What's the purpose of the lfs_config.sync callback function?

Looking through the emubd example does not seem to do anything with the data?

geky, 2018-03-07:
Good question. The sync callback is useful for data ordering. If your storage device has a large amount of caching, it may decide to write to blocks out of order, which breaks littlefs's ability to handle power failures.

Most embedded storage doesn't have much caching (such as SPI NOR flash). In these cases sync can just be an empty function that returns 0.

geky, 2018-03-17:
I think this was closed with your pr (https://github.com/geky/littlefs/pull/37), feel free to reopen if I'm wrong

closed 2018-03-17 by geky