littlefs-project/littlefs #36: Bad Block detection & Wear Leveling

jrast, 2018-03-07:
How is the wear leveling and bad block detection done?

Is it up to the lfs_conifg.prog / lfs_config.erase function to return a `LFS_ERR_CORRUPT` error to inform littlefs of a bad block or does littlefs the check by itself by reading back the data for verification?

geky, 2018-03-07:
Hi @jrast, good question. `LFS_ERR_CORRUPT` is an optional shortcut if the storage device already knows the block is bad.

littlefs verifies all writes by reading the data back. This is another case where you may want to implement the sync callback ([related](https://github.com/geky/littlefs/issues/35)) if caching would otherwise prevent the data from being read from actual storage.

Originally, I did rely on block devices returning `LFS_ERR_CORRUPT`. But I quickly learned this was a porting nightmare for users. Let me know if you see any out of date documentation or think the wording needs to be changed!

geky, 2018-03-17:
I think this was closed with your pr (https://github.com/geky/littlefs/pull/37), feel free to reopen if I'm wrong

closed 2018-03-17 by geky