littlefs-project/littlefs #39: invalid use of incomplete type ‘struct lfs_disk_dir’ and other instances.

kgillespieatmosphere, 2018-03-16:
I am attempting to utilize littlefs in my project, but it seems there are times when incomplete struct types are being used. The problem is they're being declared in a nested struct definition and apparently my flavor or version of embedded GCC tool chain isn't too happy about it. The fix is fairly simple for example, to fix the mentioned issue above we can simply change the following,

```
typedef struct lfs_dir {
    struct lfs_dir *next;
    lfs_block_t pair[2];
    lfs_off_t off;

    lfs_block_t head[2];
    lfs_off_t pos;

    struct lfs_disk_dir {
        uint32_t rev;
        lfs_size_t size;
        lfs_block_t tail[2];
    } d;
} lfs_dir_t;
```
to
```
typedef struct lfs_disk_dir {
    uint32_t rev;
    lfs_size_t size;
    lfs_block_t tail[2];
} lfs_disk_dir_t;

typedef struct lfs_dir {
    struct lfs_dir *next;
    lfs_block_t pair[2];
    lfs_off_t off;

    lfs_block_t head[2];
    lfs_off_t pos;
    
    lfs_disk_dir_t d;
} lfs_dir_t;
```

And all is well. I am going to be doing this myself for my project, but would prefer to see it also applied upstream as well. There are several other places that this occurs and I can't see why declaring the structs this way would be an issue, though I could be missing something here. Let me know what you think.

kgillespieatmosphere, 2018-03-16:
I guess I could bundle all of my "issues" under the heading of C89 support.

geky, 2018-03-17:
Hi @kgillespieatmosphere, thanks for raising an issue.

Sorry, but littlefs is decidedly C99. Features such as designated initializers, inline functions, loop initializers, `//` comments, are just too convenient and heavily used in the library.

Out of curiosity, what environment are you in that prevents C99?