littlefs-project/littlefs #40: lfs.c:638:24: error: taking address of temporary array and other instances

kgillespieatmosphere, 2018-03-16:
To resolve this problem for my project I simply moved the declaration of the temporary struct out of the function call and then used it like so,

```
    struct lfs_region region[] = {
        {entry->off, sizeof(entry->d), &entry->d, sizeof(entry->d)},
        {entry->off+sizeof(entry->d), entry->d.nlen, data, entry->d.nlen}
    };
    
    int err = lfs_dir_commit(lfs, dir, region, data ? 2 : 1);
```

This happens in several other places

```
lfs.c:654:21: error: taking address of temporary array
lfs.c:674:21: error: taking address of temporary array
lfs.c:714:13: error: taking address of temporary array
lfs.c:1119:63: warning: taking address of temporary
```

What's interesting is sometimes I just get a warning for it and others it's an error even though it all gets the same flags.

geky, 2018-03-17:
I think this is related to https://github.com/geky/littlefs/issues/39 (C89) but correct me if I'm wrong.

The disparity between warnings and errors may be because some of them are const?

kgillespieatmosphere, 2018-03-22:
Yeah it's definitely a C89 vs C99 issue. I managed to rework LittleFS to work in C89 though not sure if that is a goal for your project.

I always prefer writing against C89 because sometimes you run into that one tool chain for a project that doesn't support C99 but I would understand if your project makes C99 a minimum requirement there aren't many embedded SDK's that don't do C99 now a days.