littlefs-project/littlefs #41: Read cache internal usage

roceh, 2018-03-20:
For one of our use cases we have a file that we only read from in a very random manner (its graphical data), typically we have a table at the start with information on width, height of icons and offset into the file of the icon. For best performance we have found that setting a read cache of 1 (i.e. disabled) gives the best performance.

However we have found that some internal aspects of littlefs slow down significantly when using a cache of this size (e.g. opening and closing files). Specifically the lfs_cache_cmp and lfs_cache_crc functions slow down without a read cache.  If we tweak these functions to use a separate cache we get good performance for our data, and good open and close performance (e.g. 250ms to open and close 100 files goes down to 25ms).

It might be useful if this internal read cache size could be configurable in the standard release if you feel it doesn't over complicate things.

geky, 2018-03-20:
Ah, that is interesting! Thanks for raising the issue, this feedback is really helpful.

Originally, I thought that you would always see a performance improvement as you increased your cache size, since you would have a better chance to reuse read operations. But it turns out this was a mistake. Larger caches == larger read operations, and for linked lists this can turn into reading a large amount of useless data. This is especially notable on seek operations and is probably what you're seeing.

Unfortunately right now, there's no difference between the physical read size and the cache size, so first we'll need to add a separate cache size configuration option.

But I am planning on add something to fix this when I get the chance.

roceh, 2018-03-21:
I think it would be nice to have two separately defined read caches - one for the internal operations and one for the user operations.  Mainly to help with lfs_cache_cmp and lfs_cache_crc as currently they will read a byte at a time, even though they know the block size being crc'd.

geky, 2018-04-06:
The trade off there is code size vs runtime, the byte-based loops relying on the general purpose caching layer may be slower but they also require less code to implement.

When you also consider block devices where reads are bigger than 1 byte and need to be aligned appropriately, the caching can get quite complicated.

(But to be fair I haven't tried out any alternatives, I'm open to being proven wrong)