littlefs-project/littlefs #44: In-situ testing method?

apullin, 2018-03-26:
Great project. I am looking forward to using it extensively once I get it working.

Can you recommend a method by which I can do an exhaustive in-situ test on the hardware target for my implementation?
I am still in the debugging phase (getting extremely large block numbers), but I have to manually construct cases to comb for errors.

I don't quite see how to easily map the test suite into a single monolithic embedded application that I can run to do all the same exercises. But that would be a pretty useful thing to have ...

apullin, 2018-03-30:
Okey dokey, here's my crummy attempt at it:
https://github.com/apullin/littlefs/blob/monolithic_test/tests/monolithic.sh

which then requires `./tests/test.py << pre_gen_tests.txt`, then manually moving the contents of the C file.

geky, 2018-04-06:
This is a great idea!

I wonder if we can make it a python script that parses the test_blah.sh files.

apullin, 2018-04-06:
I am sure we can ... but maybe a less custom unit testing framework could be used, and everything could be ported to that? 

Although, the mechanics and reality of unit testing for embedded targets is still a mystery to me.
But there must be a solution.

geky, 2018-04-07:
Also I just noticed you're injecting all of the tests in your monolithic script. I don't think you want to run all of the tests since some of them are quite dependent on the testing configuration. Sorry I didn't comment earlier (I wasn't able to access github this past week).

Tests worth running on a device:
- test_dirs
- test_files
- test_seek
- test_truncate
- test_parallel - note this is not multithreaded and I should change the name
- test_paths

Tests not worth running on a device:
- test_format - depends on being able to simulate corrupt blocks
- test_alloc - depends on the exact size of the block device to create OOM situations, this should be taken from the config but it currently doesn't work well
- test_orphan - depends on being able to simulate corrupt blocks
- test_move - depends on being able to simulate corrupt blocks
- test_corrupt - depends on being able to simulate corrupt blocks

sensslen, 2018-07-17:
As far as Unit testing is concerned. We're using CppuTest for development of our embedded System quite successfully. 
When used in combination with fff (a function mocking framework) it allows for very easily understandable unit tests.

Regards, Simon

apullin, 2018-07-17:
Oh, great, sounds good. I was looking at Unity, since it seems pretty feature rich.

I don't believe function mocking is needed for this test setup? Just to port it from the custom way it is implemented now. In something that is plain C, or C and the test metas/macros.
Then set up a general test environment to pull in the user's implementation, so the test suite(s) can be run on an embedded target.