littlefs-project/littlefs #45: Available space

davidefer, 2018-03-28:
Hi
first of all thank you very much for having developed littlefs.

I have a question concerning the available space on the storage: how to calculate it?
Thank you in advance.

guillaumerems, 2018-03-28:
Hi davidefer,
Here is my implementation of a df function:

```
static int _traverse_df_cb(void *p, lfs_block_t block){
	uint32_t *nb = p;
	*nb += 1;
	return 0;
}
static int _df(void){
	int err;

	uint32_t _df_nballocatedblock = 0;
	err = lfs_traverse(&_lfs, _traverse_df_cb, &_df_nballocatedblock);
	if(err < 0){
		return err;
	}

	uint32_t available = _lfs_cfg.block_count*BLOCK_SIZE- _df_nballocatedblock*BLOCK_SIZE;

	return available;
}
```

It counts the allocated block then as we know the number of total block we can know how many block are available

davidefer, 2018-03-29:
Thank you guillaumerems, it works.
Do you know if it would be possible to get the amount of bad blocks (blocks which have reported an error while writing/erasing)?

davidefer, 2018-03-29:
Hi
I'm working with an SPI flash with 126 usable/erasable blocks of 65536 bytes each, The page write size is 256.  littlefs is V1.3.
I'm experiencing a problem while writing big files.

When I format the storage, I get the following:
Total space 8257536
Used space 262144
Free space 7995392
Where Used space = number of blocks(126) * blocks size(64KB) - free space (calculated as suggested above)

Now I write 4 files, 1MB each, in the sequence: file3, file2, file1, file0 (opened with LFS_O_WRONLY | LFS_O_CREAT | LFS_O_TRUNC)

I get the following after writing:
Total space 8257536
Used space 4718592
Free space 3538944

So far so good.

Now I write 5 files, again 1MB each, in the sequence: file4,file3,file2,file1,file0 ( file4 at the beginning and then the same as before).

Now I get the following LFS_WARN while writing file1:
        // check if we have looked at all blocks since last ack
        if (lfs->free.off == lfs->free.ack - lfs->free.begin) {
            LFS_WARN("No more free space %d", lfs->free.off + lfs->free.begin);
            return LFS_ERR_NOSPC;
        }
lfs warn:299: No more free space 251

The call stack is the following:
lfs_alloc line 299
lfs_ctz_extend line 1149
lfs_file_write line 1655

lfs->free.begin = 192
lfs->free.size=59
lfs->free.off=59
lfs->free.ack=251

If I force the file1 to be closed, then I get the following:
Total space 8257536
Used space 5832704
Free space 2424832

Please note that the files are re-written without being deleted, but since I open them again with the truncate flag, I would expect that their space is immediately freed for writing.

Could you please help?
Thank you in advance.

davidefer, 2018-03-29:
Additional information:
the problem occurs after writing the last block (125). It seems that the wrap-around doesn't work as expected.

davidefer, 2018-03-29:
I've done another attempt and now the problem occurred at block 105, so I think there is something else than the wrap-around.

closed 2018-03-31 by davidefer

geky, 2018-04-07:
Thanks @guillaumerems!

It sounds like we should probably add a lfs_fs_size function or something since this feature has been requested fairly often.

@davidefer, littlefs itself doesn't actually know which blocks are bad until it tries to write to them. It doesn't store this information.

geky, 2018-04-09:
I went ahead and tacked on a `lfs_fs_size` function to this pr https://github.com/geky/littlefs/pull/48 so it should be in v1.4. Hopefully that will be easier to use.

tomvarghese, 2019-03-18:
I tried this code and it works if I mount the fs and check before and after an operation.  However, if I unmount and remount, this number appears to reset to the original value before I did the operation.  Am I missing something simple?

geky, 2019-04-01:
@tomvarghese, that should not happen. Is it possible to create a small program that reproduces this? Thanks.

tomvarghese, 2019-04-01:
I just found the problem yesterday evening.  It was a bug in my code, was not closing the file before unmounting.

Thanks for your response though.  Really appreciate it.

Tom