littlefs-project/littlefs #47: Code is written for C99

sn00pster, 2018-04-06:
Hi,

I'm just back porting littlefs into one of our products and have hit a problem at the first hurdle, littlefs requires C99.

for example, variables are declared mid block or at the start of loops:

```
rcache->block = block;
        rcache->off = off - (off % lfs->cfg->read_size);
        int err = lfs->cfg->read(lfs->cfg, rcache->block,
                rcache->off, rcache->buffer, lfs->cfg->read_size);
```

```
static int lfs_cache_crc(lfs_t *lfs, lfs_cache_t *rcache,
        const lfs_cache_t *pcache, lfs_block_t block,
        lfs_off_t off, lfs_size_t size, uint32_t *crc) {
    for (lfs_off_t i = 0; i < size; i++) {
```

should really be:

```
static int lfs_cache_crc(lfs_t *lfs, lfs_cache_t *rcache,
        const lfs_cache_t *pcache, lfs_block_t block,
        lfs_off_t off, lfs_size_t size, uint32_t *crc) {
    lfs_off_t i = 0;
    for (i = 0; i < size; i++) {
```

It's obviously easy enough to fix and doesn't do any harm, just makes it more compatible with more compilers.

Also technically comments should be in /** **/ and not // for the same reason (although I've yet to come across a compiler that doesn't like //

sn00pster, 2018-04-06:
This is also a problem (but easy enough to fix):

`    const lfs_block_t tpair[2] = {pair[0], pair[1]};`

`        if (lfs_paircmp(oldpair, (const lfs_block_t[2]){0, 1}) == 0) {`

sn00pster, 2018-04-07:
The compiler I’m using doesn’t support c99!

I completed most of the changes, I know it’s down to coding styles but some of the places you’ve used named initialisers actually creates more lines of code than not using them!

Once I’ve completed my changes, I’ll clone to the repo copy my files over the top and go through the git changes to make sure I didn’t mess anything up.

geky, 2018-04-09:
I'm just curious which compiler it is.

But no, you're right (and @kgillespieatmosphere). It'd be a shame if other had to duplicate your efforts.

I'd be happy to host a c89 version, but I would prefer to treat it as a port to a different language.

If you have changes that compile in c89, feel free to create a pr. I'm thinking I'll create it's own repo and link to it from the README.

sn00pster, 2018-04-09:
Huh, after I'd finished making the last change I realised there was a compiler option to turn on c99 compatibility!

Hangs head.....

geky, 2018-04-09:
Haha, well that's an easy solution. Offer still stands to create a branch for C89 if anyone wants to create a pr for it.

sn00pster, 2018-07-18:
@geky I'm happy to close this one, I agree it's a non-issue and keeping up to date between a C89 and C99 variant would be a waste of time and effort.

geky, 2018-07-18:
Thanks! Sorry there's a lot of lingering open issues. I need to go through all of them but haven't had a good chance to yet.