littlefs-project/littlefs #49: Is LFS_NAME_MAX 255 bytes a little larger?

xjtuecho, 2018-04-13:
Most files or directories do not have a so long name.  If user define 'struct lfs_info' on the stack, it may be unsafe. I have reduced LFS_NAME_MAX  to 32 and it works. But I don't want to modify ‘lfs.h’. Should  this macro appear in ‘lfs_util.h’ ?

geky, 2018-04-18:
You're right! I've actually added an ifdef so you can redefine it to a good value is good for your system: https://github.com/geky/littlefs/pull/48

One concern is portability (ie mounting a storage device onto a PC), since different systems may support different name sizes. In #48 I added the file name size as an entry in the superblock at format time, so other systems should respect it.

That's the main reason it's not supported quite yet.

geky, 2018-04-29:
No, not on windows at the moment. Though I hope put one together when I have the time (or I'd be excited if someone else put one together).

There is a fuse wrapper which allows you to mount LittleFS in linux. You could use a virtual machine, although it wouldn't be convenient:
https://github.com/geky/littlefs-fuse