littlefs-project/littlefs #50: V1.3 release conflict?

davidefer, 2018-04-17:
Hi geky
I've downloaded the V1.3 at end of March and now there is a new version 1.3 including the bugfix of the lookahead.
By the way if you look at the releases page the V1.3 is dated February the 20th.
Is there a multiple V1.3? Does it make sense to generate a V1.4?
Thanks in advance.

geky, 2018-04-18:
Hi @davidefer,

The release script is currently set up to copy any bug fixes into the most recent release. The idea is if you download a specific release you will also get any bug fixes related to the release.

The minor version number is only incremented when a feature is added, it's still backwards compatible but more likely to introduce bugs and may require additional work to update if the user wants to take advantage of new features.

If you need to specify a frozen point in history, you can use the github commit hash.

Let me know if this concerns you.

closed 2018-04-23 by davidefer

geky, 2018-07-19:
We're looking at improving this over here https://github.com/ARMmbed/littlefs/issues/79