littlefs-project/littlefs #52: limitation when no dynamic memory allocation is used

husigeza, 2018-04-28:
Hello,

Could you help me regarding what limitations exist when statically allocated buffers are passed to the configuration structure instead of using dynamic memory allocation ?

Above the "file_buffer" it writes: "If enabled, only one file may be opened at a time." -> I could verify the problem here (when writing both files, the content of the second writing will be written into the first file too), however I would have expected that the second file open returns a negative error code.

Thanks,
Geza

geky, 2018-04-29:
Ah! good idea. I just didn't think of returning an error code in that condition.

Does this fix look good to you? (returns LFS_ERR_NOMEM / -ENOMEM)
https://github.com/geky/littlefs/compare/fix-multi-static-file

kewaldeshpande, 2019-06-04:
@husigeza without dynamic memory allocation , are you saying we  can still use LittleFS with static buffers ? I am trying to use littleFS for my STM where I dont have dynamic memory.

husigeza, 2019-06-04:
@kewld yes, it is possible, however there are limitations, such as you can only have 1 file opened at a time.

FreddieChopin, 2019-06-05:
> @kewld yes, it is possible, however there are limitations, such as you can only have 1 file opened at a time.

Not true.

> I am trying to use littleFS for my STM where I dont have dynamic memory.

You have "dynamic memory" - just use it.

geky, 2019-06-06:
> > @kewld yes, it is possible, however there are limitations, such as you can only have 1 file opened at a time.
>
> Not true.

This changed in v1.5: https://github.com/ARMmbed/littlefs/pull/58

It was true when this issue was created :) Which means I've probably left this issue open for too long. Feel free to reopen if there's still an issue.

> are you saying we can still use LittleFS with static buffers ?

You can specify a buffer per file using `lfs_file_opencfg`:
``` c
// must be equal to cache size in lfs config
uint8_t file_buffer[config.cache_size];
// note we must zero initialize unused fields for backwards compatibility
// designated initializers do this for us
struct lfs_file_config file_config = {
    .buffer = file_buffer;
};
// open file
lfs_file_t file;
int err = lfs_file_opencfg(&lfs, &file, "path/to/file",
        LFS_O_WRONLY | LFS_O_CREAT, &file_config);
assert(!err);
```

nickray, 2019-06-07:
> You have "dynamic memory" - just use it.

Not using heap allocations is a legitimate use case - not everyone splashes around with an RTOS :yum:.

I'm happy to report I just got LittleFS to run via Rust bindings on an STM32L432 microcontroller - no heap allocations, no libc. I only had to patch in some <string.h> implementations.

FreddieChopin, 2019-06-07:
> Not using heap allocations is a legitimate use case - not everyone splashes around with an RTOS

Using or not using dynamic allocations have nothing to do with a RTOS.

geky, 2019-06-10:
> I'm happy to report I just got LittleFS to run via Rust bindings on an STM32L432 microcontroller

Cool! Is this with the littlefs crate I've seen floating around?
https://docs.rs/littlefs/0.2.0/littlefs/

> Which means I've probably left this issue open for too long. Feel free to reopen if there's still an issue.

I wrote that then forgot to actually close this! closing... (discussion can still continue)

closed 2019-06-10 by geky

nickray, 2019-06-11:
> Cool! Is this with the littlefs crate I've seen floating around?
> https://docs.rs/littlefs/0.2.0/littlefs/

Yeah, I bumped it to v2 and had to [patch it a little](https://github.com/nickray/rust-littlefs/commits/master), looking forward to using file attributes which should come in handy. [Implementing the `Storage` trait](https://github.com/nickray/zissou/blob/20f7aa1ac517bf6b81fe618d90e7eb813ebc8ce8/src/gordon.rs#L18-L37) felt really really nice and elegant, I'm in a honeymoon/fanboy phase with your `littlefs` currently!