littlefs-project/littlefs #54: Problems with the size of the file to read

Baradablr, 2018-05-15:
Hello everybody! I have a problem with the file system with the following settings:

    .read_size = 256,
    .prog_size = 256,
    .block_size = 4096,
    .block_count = 7168, // - 1024 for other needs
    .lookahead = 1024

With these settings, the file system can not normally read the recorded files larger than 2 blocks. After analyzing the code, I found a problematic place in the file lfs.c:

1122 int err = lfs_cache_read (lfs, rcache, pcache, head, 4 * skip, & head, 4);

After this line of code, I have the wrong address of the block header. Further, I did not go into depth, I need more knowledge in the file system device. Please help. =)

geky, 2018-05-17:
Hi @Baradablr, sorry for the delay. I tried reproducing the error, but on my device it seems to be working fine.

Exactly how many bytes large is the file write that fails? Or do you have a code snippet you could share that demonstrates the problem?

What address did the cache_read load? It should be iterating over the linked list that is the file at that point, so "head" should be updated with the address of each block in the chain.

Baradablr, 2018-05-18:
`static int lfs_ctz_find(lfs_t *lfs,
        lfs_cache_t *rcache, const lfs_cache_t *pcache,
        lfs_block_t head, lfs_size_t size,
        lfs_size_t pos, lfs_block_t *block, lfs_off_t *off) {
    if (size == 0) {
        *block = 0xffffffff;
        *off = 0;
        return 0;
    }

    lfs_off_t current = lfs_ctz_index(lfs, &(lfs_off_t){size-1});
    lfs_off_t target = lfs_ctz_index(lfs, &pos);

    while (current > target) {
        lfs_size_t skip = lfs_min(
                lfs_npw2(current-target+1) - 1,
                lfs_ctz(current));

        int err = lfs_cache_read(lfs, rcache, pcache, head, 0, &head, 4);
        head = lfs_fromle32(head);
        if (err) {
            return err;
        }

        LFS_ASSERT(head >= 2 && head <= lfs->cfg->block_count);
        current -= 1;
    }

    *block = head;
    *off = pos;
    return 0;
}`
This changes make my fs working. I think that it is necessary to remove the offset at each iteration, because the block address is always on offset 0. I also removed the offset at each iteration of the block search. With such fixes, the fs regularly writes and reads files larger than 5 MB.

geky, 2018-05-18:
That's a good workaround, but it doesn't take advantage of the CTZ skip-list. Do you know the exact size of the file that caused the problem? I'd be interested in fixing this bug in the general case.