littlefs-project/littlefs #59: Change in license

sg-, 2018-06-11:
Interest in littlefs is growing and we’d like to enable users to combine it with other open source projects, regardless of license. Unfortunately, due to the [Free Software Foundation considering Apache-2.0 to be incompatible with GPL-2.0](https://www.gnu.org/licenses/license-list.en.html#apache2), this creates a limitation for some open source projects. 

In order to continue the growth of littlefs and enable friction-less combination with other open source projects, we are seeking to change the license for littlefs from Apache-2.0 to [BSD-3-Clause](https://spdx.org/licenses/BSD-3-Clause.html). We have emailed all contributors to request permission to change the license. If you have been a contributor and received an email, please help us continue the development of littlefs by responding! Future contributions will be based on inbound = outbound BSD-3-Clause and pending contributions will need to be rebased on master after the license change has taken place to enact the appropriate inboud=outbound license.

Thanks for all the support and interest in littlefs 👍

jlovejoy, 2018-06-15:
@aldot , @jrast , @Sim4n6 - as previous contributors to littlefs, can you please provide your permission to re-license your contributions under BSD-3-Clause? (in the spirit of openness and transparency, we are asking all contributors, no matter the contribution to affirm).  Thanks!

Sim4n6, 2018-06-15:
Ok go ahead

closed 2018-06-21 by geky

pabigot, 2019-07-23:
The current v2 `lfs.c` is Apache-2.0 licensed at the top of the file.  Maybe the change got lost in the version rev.

geky, 2019-07-23:
Ah! That is incorrect. Thanks for point it out, I will fix it. The license is still BSD-3-Clause.

geky, 2019-07-23:
https://github.com/ARMmbed/littlefs/pull/240