littlefs-project/littlefs #63: Supporting fstat (statting an already open file)

rojer, 2018-06-25:
It is a fairly common pattern to open a file, _then_ stat it to determine what the size of the file that i just opened.
How hard would it be to implement?
I see that size is readily available in the file_t, not sure how hard would it be to get the current name of the file.

geky, 2018-06-25:
Hi @rojer

Does the `lfs_file_size` function work for you?
``` c
// Return the size of the file
//
// Similar to lfs_file_seek(lfs, file, 0, LFS_SEEK_END)
// Returns the size of the file, or a negative error code on failure.
lfs_soff_t lfs_file_size(lfs_t *lfs, lfs_file_t *file);
```
https://github.com/ARMmbed/littlefs/blob/master/lfs.h#L388-L392

Here's it's use in the FUSE wrapper:
https://github.com/ARMmbed/littlefs-fuse/blob/master/lfs_fuse.c#L232-L235

rojer, 2018-06-25:
yeah, looks like that should do it. thanks!

closed 2018-06-25 by rojer