littlefs-project/littlefs #64: Unique file identifier

rojer, 2018-06-25:
What can be used as a unique file identifier of a file on a filesystem? Basically, to set the `st_ino` field of the `struct stat`.

geky, 2018-06-25:
Oh, that's a bit difficult. You would need at least 10 bytes to represent a file location uniquely. Does that fit in the `st_ino` field?

On disk, a file is a single metadata entry in it's parent's directory pair, and a directory pair is two block addresses. There's not a single block that is the file's data.

What functionality does the `st_ino` field need to accomplish? If it's just needed for uniquity, could you assign the `st_ino` field an incrementing number every time a file is allocated?

blomnik, 2020-08-17:
Hello. This will be convinient to lock access (with mutex) per file, not locking whole FS. Can you direct me to determine unique portion of data in LFS which can be used as key in hash-table of opened files?

geky, 2020-08-18:
In the most recent version you need an id (16-bits) in a metadata pair (2x32-bits).

Also, this isn't exposed in a proper API given the limited usefulness, so it's susceptible to change in future version.

Consider using the file path instead. It's annoying that the file path is an unbounded piece of info, but the file path _is_ intended to be a unique identifier in the filesystem.

The pieces of info can be found here, but internal struct variables aren't guaranteed to be consistent between version. This may break in the future. But I'll add it here for the info:
``` c
lfs_file_t file;
file.m.pair  // uint32_t[2] used to find the pair of blocks where the file's metadata is stored
file.id      // uint16_t id used to find the file in the metadata
```

If this info is valuable to users, we can add the info as a part of the `lfs_stat` function, but currently the usefulness seems limited.