littlefs-project/littlefs #68: Corruption

sn00pster, 2018-07-06:
Managed to get a corruption, no idea how, but if. I attempt to open this file for read/write I get the following output.  I let the GSM update the time and left it a few minutes and it had no problem opening a file in a different location.  (All the ini ones are it happily opening the configuration file at boot)

opened file system.ini 200146bc
closed file 200146bc
opened file system.ini 200146bc
closed file 200146bc
opened file system.ini 20009a5c
closed file 20009a5c
opened file /default.amx 20009a5c
reading 60
reading 11704
closed file 20009a5c
opened file app.ini 20009964
closed file 20009964
opened file app.ini 20009964
closed file 20009964
opened file app.ini 20009964
closed file 20009964
opened file app.ini 20009964
closed file 20009964
opened file app.ini 20009964
closed file 20009964
opened file app.ini 20009964
closed file 20009964
opened file app.ini 20009964
closed file 20009964
opened file app.ini 20009964
closed file 20009964
opened file app.ini 20009964
closed file 20009964
opened file app.ini 20009964
closed file 20009964
opened file app.ini 20009964
closed file 20009964
opened file app.ini 20009964
closed file 20009964
opened file app.ini 20009964
closed file 20009964
opened file app.ini 20009964
closed file 20009964
opened file system.ini 2000992c
closed file 2000992c

opened file /logs/2013/01/01.k2a 20014658
lfs error:469: Corrupted dir pair at 95 94
opened file /logs/2013/01/01.k2a 20014658
lfs error:469: Corrupted dir pair at 95 94
opened file /logs/2018/07/06.k2a 20014658
closed file 20014658

I know I'm presenting a dead parrot here with no information, this is exactly the type of bug report I hate getting from a user and just shrug my shoulders at them!

geky, 2018-07-06:
> I know I'm presenting a dead parrot here with no information, this is exactly the type of bug report I hate getting from a user and just shrug my shoulders at them!

Haha, thanks for posting an issue though. It's helpful if other users run into a similar situation.

What is the hash of the version of littlefs you are usings? `git log`

Is there any way you are able to get a binary dump of the storage? That would help immensly. You can email it to me directly if you don't want to share it publically.

sn00pster, 2018-07-09:
Unforuntately something else came up and I had to revert back to an earlier revision of firmware to do some development fixes on and that uses a FAT filing system, so I had to reformat it.

This does beg the question, if little fs when accessing a file/dir knows that both are invalid/corrupt, then should it fix it? (i.e delete the file), I don't know what happens, but I'm assuming that I cannot write to the file with both blocks corrupted? Or does a write do exactly what I think it should do in this circumstance and fix the entry?

geky, 2018-07-10:
That's ok, let us know if it pops up again.

Currently littlefs only checks the validity of metadata, which is stored in the directory block. If a file is corrupt, that means its parent directory is corrupt. And it's likely that neighboring files in that directory will also be corrupt.

At this point you would need to delete and recreate the parent directory. Unfortunately, the way the `lfs_remove` function is written right now, that would actually be impossible, since it tries to fetch the directory to check if it is empty or not, which would fail since it is corrupted (related https://github.com/ARMmbed/littlefs/issues/43).

---

The directory blocks aren't completely vulnerable. They are actually pairs, in which one block always contains a backup of the directory's data. If one directory block is corrupt, the other is picked up implicitly. So this error can only happen if both directory blocks are corrupt.

To get to this sort of corruption, most likely something outside of a directory affected the directory blocks. I'm guessing there's a bug where littlefs is accidentally writing over the directory blocks as a part of a different operation, but it's hard to know.

avtolstoy, 2018-08-28:
I think we are running into a similar issue under unknown circumstances on some of our devices: `lfs error: Corrupted dir pair at 4 2`. We are using half of MX25L3233F 32Mb QSPI flash with the following settings:
```
read_size = 256,
prog_size = 256,
block_size = 4096,
block_count = 512,
lookahead = 128
```

I am attaching a dump affected by this: [exflash.bin.gz](https://github.com/ARMmbed/littlefs/files/2328387/exflash.bin.gz)

We are using littlefs v1.6.0.

I do really hope this is just simply a bug in our QSPI driver, but if it's not, I hope that the dump at least sheds some light on how we end up in this corrupted state. Thanks!

guillaumerems, 2018-10-09:
Hi @geky ,

I am using littlefs in my product since the beginning of little and I encounter some filesystem corruption, hopefully not so often.
We do not suspect a power loss issue but more a high write file rate.

When the filesystem is corrupted, what is strange is that the read files is still working properly, but only no write are no more possible.
We have a df function to check periodically the available free space and this function also fails as it calls lfs_traverse.

After diagnosis, we observe that the failure come from lfs_dir_fetch function with a wrong crc calculation. And LFS tries to read a block header at a totally wrong address, because the block address in the pair is not correct. (for example 137035812)
We try to found a way to catch this issue and repair the filesystem when it happens, I do not care to lose a file but I do not want to lose the entire filesystem when it happens.

Maybe it could be great to check the pair integrity (check maximum block, ...) ? or other solution ?

And is it possible to remove a bad block when we found one ?

LaurentLouf, 2019-06-26:
I also run into some corruptions, I get the following output : 

```
E (60888) LittleFS: file_open(0x3fff2240, "/night_data/1510196793/sleep_duration_minute", 0x8000) -> -2
I (60888) SPIFileSystem: Start reading from night 1510196793
E (60928) LittleFS: file_open(0x3f410970, "/night_data/1510196793/accelero", 0x8000) -> -2
E (60958) LittleFS: file_open(0x3f410970, "/night_data/1510196793/water", 0x8000) -> -2
E (60998) LittleFS: file_open(0x3f410970, "/night_data/1510196793/heatsink", 0x8000) -> -2
E (61068) LittleFS: file_open(0x3fff2240, "/night_data/1510196793/light", 0x8000) -> -2
E (61108) LittleFS: file_open(0x3fff2240, "/night_data/1510196793/ambient_ht", 0x8000) -> -2
I (61108) Main: New sleep data read from flash, updating characteristic value
I (61148) SPIFileSystem: All data read from flash for night 1510196793, removing it from flash
E (71528) lfs: dir_remove
I (71528) SPIFileSystem: Remove /night_data/1510196793/night_settings from the file system returned -4001
E (71528) LittleFS: Corrupted dir pair at 3 2
E (71528) lfs: dir_find
I (71538) SPIFileSystem: Remove /night_data/1510196793/bed_time from the file system returned -138
E (71548) LittleFS: Corrupted dir pair at 3 2
E (71548) lfs: dir_find
I (71548) SPIFileSystem: Remove /night_data/1510196793 from the file system returned -138
E (71558) LittleFS: Corrupted dir pair at 3 2
E (71558) LittleFS: dir_open(0x3fff2240, "/night_data") -> -138
E (71568) SPIFileSystem: Error trying to fetch the timestamp of the next night
E (71578) LittleFS: Corrupted dir pair at 3 2
E (71578) LittleFS: dir_open(0, "/night_data") -> -138
E (71578) SPIFileSystem: Error trying to fetch the timestamp of the next night
```

Funny thing is, if I reboot the system and try to perform the same operations (which should fail if there is indeed a corruption), I get exactly the same log. The operations are basically : 
- get through the "night_data" directory to fetch the first timestamp, that corresponds to a directory containing a bunch of files
- read the data of the existing files and fill a buffer with the data
- delete the files and the directory
- repeat

Any idea what is happening here ? Could the remove cause some corruption ?