littlefs-project/littlefs #78: Sudden drop in performance when fs is mounted for a long time

FreddieChopin, 2018-07-17:
I've already mentioned this in #75 , but let's move this to another thread.

On my fs I have just 3 files - 2 very small, and the third one is pretty big (1 or 2 MB). I may also try to see whether the same thing happens when the fs is initially completely empty. The fs is mounted on a SD HC 32 GB card and uses standard settings (512 for all block sizes, 512 for lookahead).

The test procedure goes like this:
1. mount fs
2. check whether there's a file called `v.txt`, if not - create it and save string "0" in there.
3. in an infinite loop:
3.a. open file `v.txt`
3.b. read its contents with `scanf()` into a variable
3.c. rewind the file
3.d. increment the variable
3.e. `printf()` the value of variable (as a string) into the file
3.f. close the file

I've also added some measurements, which check how many times the loop gets repeated per second. Disregarding occasional allocator scans (which take ~7 seconds), the loop performance is mostly constant and around 20-22 times per second. However once every few thousand writes, the performance just drops instantly to 3-4 times per second.

Let's start with a plot (;

[image]

(x axis - value of the counter read, incremented and written in the loop, y axis - loops per second)

Of course the small "drops" are caused by allocator scan, just disregard these.

In the plot there are three "drops", however the middle one (at 71580) was too short and is not visible here. The "drop" in performance does _NOT_ disappear by itself. In the code I've added a `volatile bool` flag, which I can set via debugger, which causes the fs to be unmounted and mounted again. After such operation the performance is instantly back at 20-22 loops per second (disregarding initial allocator scan, which takes over a minute the first time it is run). This is most likely not related to SD card FTL, as when the "drop" occurs I can pause my code at the beginning of the loop (SD card is therefore not busy and not doing anything) for as long as I like - after I resume it, the performance does not change.

Some numbers:
- first drop after 12220 reads, rewinds and writes,
- second drop after 12200 loops,
- third drop after 12207 loops.

The numbers are pretty consistent here, so it seems the value 12200 has something to do with it.

The last "drop" is the longest one, I was wondering whether anything more happens after some time, but I got bored waiting (; No change observed.

I realize my test is a bit overly complex (use of `scanf()` and `printf()`), but at the same time I was testing some of the features I implemented in my [C++ RTOS - distortos](http://distortos.org/). Exactly the same thing was observed when I used the simpler approach of using raw reads and writes. That's why I think the problem is easily reproducible, just do _NOT_ unmount the fs after each iteration.

Let me know what else can I do to assist you in finding any possible problem here. If you want the dump of some fs-related variable, whole RAM or the fs itself - just let me know.

Here's my test code for reference:

```
int main()
{
 distortos::devices::SpiSdMmcCard card {distortos::board::spi3Master, slaveSelect, 5000000 / 2};
 distortos::LittlefsFileSystem fileSystem;

 volatile bool flag {};

 {
  const auto ret = fileSystem.mount(card);
  if (ret != 0)
  {
   fiprintf(debugStream, "fileSystem.mount(): %d\r\n", ret);
  }
 }
 {
  const auto ret = fileSystem.getFileStatus("v.txt");
  if (ret.first != 0)
  {
   fiprintf(debugStream, "fileSystem.getFileStatus(): %d\r\n", ret.first);
  }
  if (ret.first == ENOENT || ret.second.st_size == 0)
  {
   const auto ret2 = distortos::openFile(fileSystem, "v.txt", "w");
   if (ret2.first != 0 || ret2.second == nullptr)
   {
    fiprintf(debugStream, "distortos::openFile(\"w\"): %d\r\n", ret.first);
   }
   const auto ret3 = fiprintf(ret2.second, "0");
   if (ret3 != sizeof("0"))
   {
    fiprintf(debugStream, "fiprintf(ret2.second, \"0\"): %d e %d\r\n", ret3, errno);
   }
   const auto ret4 = fclose(ret2.second);
   if (ret4 != 0)
   {
    fiprintf(debugStream, "fclose(ret2.second): %d e %d\r\n", ret4, errno);
   }
  }
 }

 unsigned int count {};
 std::chrono::seconds time {};
 uint64_t v;

 while (1)
 {
  const auto now = std::chrono::duration_cast<std::chrono::seconds>(distortos::TickClock::now().time_since_epoch());
  if (time != now)
  {
   fiprintf(debugStream, "%u %" PRIu64 "\r\n", count, v);
   time = now;
   count = {};
  }

  if (flag == true)
  {
   flag = {};
   {
    const auto ret = fileSystem.unmount();
    if (ret != 0)
    {
     fiprintf(debugStream, "fileSystem.unmount(): %d\r\n", ret);
     distortos::ThisThread::sleepFor(std::chrono::seconds{10});
    }
   }
   {
    const auto ret = fileSystem.mount(card);
    if (ret != 0)
    {
     fiprintf(debugStream, "fileSystem.mount(): %d\r\n", ret);
     distortos::ThisThread::sleepFor(std::chrono::seconds{10});
    }
   }
  }

  FILE* file;
  {
   int ret;
   std::tie(ret, file) = distortos::openFile(fileSystem, "v.txt", "r+");
   if (ret != 0 || file == nullptr)
   {
    fiprintf(debugStream, "distortos::openFile(): %d\r\n", ret);
    distortos::ThisThread::sleepFor(std::chrono::seconds{10});
    continue;
   }
  }
  {
   const auto ret = fiscanf(file, "%" SCNu64, &v);
   if (ret != 1)
   {
    fiprintf(debugStream, "fiscanf(): %d e %d\r\n", ret, errno);
    distortos::ThisThread::sleepFor(std::chrono::seconds{10});
    continue;
   }
  }

  rewind(file);

  fiprintf(debugStream, "%" PRIu64 "\r\n", v);

  ++v;

  {
   const auto ret = fiprintf(file, "%" PRIu64, v);
   if (ret < 0)
   {
    fiprintf(debugStream, "fiprintf(): %d e %d\r\n", ret, errno);
    distortos::ThisThread::sleepFor(std::chrono::seconds{10});
    continue;
   }
  }
  {
   const auto ret = fclose(file);
   if (ret != 0)
   {
    fiprintf(debugStream, "fclose(): %d e %d\r\n", ret, errno);
    distortos::ThisThread::sleepFor(std::chrono::seconds{10});
    continue;
   }
  }

  ++count;
 }
}
```

FreddieChopin, 2018-07-17:
Contents of file system during above test:
```
$ ls -la
total 4
drwxrwxrwx 0 root    root          0 1970-01-01  .
drwxr-xr-x 5 freddie freddie    4096 07-11 19:43 ..
-rwxrwxrwx 0 root    root    2097152 1970-01-01  dump.bin
-rwxrwxrwx 0 root    root          8 1970-01-01  v.bin
-rwxrwxrwx 0 root    root          5 1970-01-01  v.txt
```
I'm testing with empty fs, freshly formatted.

FreddieChopin, 2018-07-17:
[image]

Empty file system, just one small file. "Drop" every 16382 writes (+/-2).

FreddieChopin, 2018-07-19:
> I was going to see whether increasing block size has any influence, but surprisingly increasing read, prog and erase block sizes to 2 kB caused the whole thing to work 3x slower (~6-7 loops per second instead of previous ~20-22), possibly because now each write/read cycle from the SD card has to use multiblock operations which contain more "busy" periods. I'll try with 1 kB, so to see the results faster.

Indeed - increasing all block sizes to 1 kB (x2) caused the problem to appear faster - after about 8187 loop cycles (note how close this value is to 0x1fff). This however doesn't make the problem appear much faster (counting in time), as the loops are slower by about 30-40%, with ~13 iterations per second instead of ~21.

I've then tried another thing - I left read and prog block sizes at 512 B and set erase block size to 2048 B (x4). This caused the problem to appear 4x earlier - after ~4095 iterations, with the speed of loops at ~22 per second before the drop. So now I have a way to make the problem appear much sooner, so it may be easier to profile what's going on and where.

FreddieChopin, 2018-07-19:
Heh, you were right. The problem is the SD card. It turns out that erasing blocks which are at 8M - 12M takes 130 ms, while erasing blocks other blocks (above or below this range) - takes just 3 ms. The problem occurs when littlefs goes into the range of these addresses. SD cards are strange... I've also noticed that further there are some slower blocks too.

The "AU" of this particular card is 4 MB, and from the description in the spec I guess that this is the real erase size of the underlying flash, hidden behind FTL. So it seems that some AUs are faster than others and this is just how life goes.

BTW - I've noticed that for each write to the file (which are placed at increasing addresses) there's one write to either block nr 1 or block nr 2. I guess these are the superblocks you mentioned in other places. Won't they get worn-out pretty quickly in that case? You cannot assume that the user will always use folders for every file (in which case I guess the superblocks for root folder wouldn't be modified).

closed 2018-07-19 by FreddieChopin

geky, 2018-07-19:
Ah whoops, so it was 8MB not 8GB, I was off by just a few orders of magnitude.

It makes sense if you consider the main target for SD cards. The FAT filesystem stores the FAT table there, so if an SD card manufacturer spends the money to make the first arena faster, the entire filesystem benefits as a result, and it's much cheaper than speeding up the entire disk.

Unfortunately, that conflicts with LittleFS's attempts to distribute wear.

> BTW - I've noticed that for each write to the file (which are placed at increasing addresses) there's one write to either block nr 1 or block nr 2.

It should actually be writing to blocks 2 and 3. Blocks 0 and 1 are the superblock and are only updated when blocks 2 and 3 (the root directory) is moved.

I do have a plan upgrade this by actually adding blocks into the superblock chain every time the root directory moves so that the wear on the superblock decreases exponentially. But as it is now you at least get an order of magnitude more writes to the root block.