littlefs-project/littlefs #87: wear leveling

barbiani, 2018-08-06:
Hello everybody,

I have just compiled littlefs with my application that uses a 32k fram.

We can see that it stays rewriting to the 5 first blocks.

Also, sometimes I get big increments in boot_count. Still debugging it.

Runing in a 16bit mcu. Had to adapt lfs_npw2, lfs_ctz, lfs_popc, lfs_scmp and lfs_fromle32.

With the configuration:

> const struct lfs_config cfg = {
>     .read  = user_provided_block_device_read,
>     .prog  = user_provided_block_device_prog,
>     .erase = user_provided_block_device_erase,
>     .sync  = user_provided_block_device_sync,
> 
>     // block device configuration
>     .read_size = 64,
>     .prog_size = 64,
>     .block_size = 1024,
>     .block_count = 32,
>     .lookahead = 1024,
> };

And code:

>         err = lfs_mount(&lfs, &cfg);
>         // reformat if we can't mount the filesystem
>         // this should only happen on the first boot
>         if (err) {
>             lfs_format(&lfs, &cfg);
>             lfs_mount(&lfs, &cfg);
>         }
>         // read current count
>         uint16_t boot_count = 0;
>         err = lfs_file_open(&lfs, &file, "boot_count", LFS_O_RDWR | LFS_O_CREAT);
>         err = lfs_file_read(&lfs, &file, &boot_count, sizeof(boot_count));
>         // update boot count
>         boot_count += 1;
>         err = lfs_file_rewind(&lfs, &file);
>         err = lfs_file_write(&lfs, &file, &boot_count, sizeof(boot_count));
>         // remember the storage is not updated until the file is closed successfully
>         err = lfs_file_close(&lfs, &file);
> #if 1
>         err = lfs_dir_open(&lfs, &dir, "/");
>         if (!err) {
>             for (;;) {
>                 lfs_dir_read(&lfs, &dir, &info);
>                 if (!info.name[0])
>                     break;
>                 console_printf("1:/%s\t\t%ld\n\r", info.name, info.size);
>             }
>             lfs_dir_close(&lfs, &dir);
>         }
> #endif
>         // release any resources we were using
>         err = lfs_unmount(&lfs);
>         // print the boot count
>         console_printf("boot_count: %d\n\r", boot_count);
> 

`> fram erase
> fram lfs

lfs debug:22: Reading from block 0 offset 0 size 64
lfs debug:22: Reading from block 1 offset 0 size 64
lfs error:497: Corrupted dir pair at 0 1
lfs error:2226: Invalid superblock at 0 1
[... 1988 lines cut ...]
lfs debug:22: Reading from block 3 offset 0 size 64
lfs debug:22: Reading from block 4 offset 0 size 64
lfs debug:22: Reading from block 0 offset 0 size 64
lfs debug:22: Reading from block 1 offset 0 size 64
lfs debug:22: Reading from block 0 offset 0 size 64
lfs debug:22: Reading from block 3 offset 0 size 64
lfs debug:22: Reading from block 2 offset 0 size 64
lfs debug:22: Reading from block 3 offset 0 size 64
lfs debug:35: Erasing block 5
lfs debug:28: Writing to block 5 offset 0 size 64
lfs debug:22: Reading from block 5 offset 0 size 64
lfs debug:22: Reading from block 3 offset 0 size 64
lfs debug:22: Reading from block 2 offset 0 size 64
lfs debug:22: Reading from block 3 offset 0 size 64
lfs debug:35: Erasing block 2
lfs debug:28: Writing to block 2 offset 0 size 64
lfs debug:22: Reading from block 2 offset 0 size 64
lfs debug:22: Reading from block 3 offset 0 size 64
lfs debug:22: Reading from block 2 offset 0 size 64

1:/.		0
1:/..		0
1:/boot_count		2
boot_count: 52
`

e107steved, 2018-08-09:
I've got lfs running OK on a FRAM (although 128K/256K); bit slow due to I2C comms, but solid. No need to modify anything in the source code.
Suggest you reduce your block size from 1024 - it can go as low as 128 bytes.

barbiani, 2018-08-09:
Yes, actually it is a FM25W256 (32 kilobytes).

Had troubles with crashes while setting small block sizes.

geky, 2018-08-09:
Hi @barbiani, sorry for the late response.

> Runing in a 16bit mcu

Oooh, that's interesting. Out of curiousity which processor/fram is this? If you're able to share.

> Had troubles with crashes (ASSERT) while setting small block sizes.

Is this the assert that is failing?
https://github.com/ARMmbed/littlefs/blob/master/lfs.c#L2077-L2079

That could also mean that `lfs_npw2` is implemented incorrectly. You can try compiling with `-DLFS_NO_INTRINSICS`, which will fall back to standard C implementations.

e107steved, 2018-08-14:
The definition I found is: int __builtin_clz (unsigned int x)  
(https://gcc.gnu.org/onlinedocs/gcc/Other-Builtins.html), so that could well not work as intended on a 16-bit processor.

geky, 2018-08-14:
That's a good point, and probably the source of the problem. I'm guessing changing it to __builtin_clzl should work?

As well as changing the others in lfs_util.h:
- __builtin_popcount -> __builtin_popcountl
- __builtin_clz -> __builtin_clzl
- __builtin_ctz -> __builtin_ctzl

I'll need to check if the other compilers support the "l" suffix before adopting this change.

barbiani, 2018-08-14:
It is a dspic33ep.

Exactly this one:
LFS_ASSERT(4*lfs_npw2(0xffffffff / (lfs->cfg->block_size-2*4)) <= lfs->cfg->block_size);

I played dump and did not notice that, under the conformal coating, the memory IC had just 8kbytes of space and I thought it had 32.

These are the intrinsic I am using at the moment.

```
// Find the next smallest power of 2 less than or equal to a
static inline uint32_t lfs_npw2(uint32_t a) {
    unsigned count = 0;
    // First a in the below condition
    // is for the case where a is 0
    if (a && !(a & (a - 1))) {
        return a;
    }
    while (a != 0)  {
        a >>= 1;
        count += 1;
    }
    return (1 << count);
}

// Count the number of trailing binary zeros in a
// lfs_ctz(0) may be undefined
static inline uint32_t lfs_ctz(uint32_t a) {
    int count = 0;
    while ((a & 1) == 0) {
        a = a >> 1;
        count++;
    }
    return (count);
}

// Count the number of binary ones in a
static inline uint32_t lfs_popc(uint32_t a) {
    unsigned int count = 0;
    while (a) {
      count += a & 1;
      a >>= 1;
    }
    return (count);
}

// Find the sequence comparison of a and b, this is the distance
// between a and b ignoring overflow
static inline int lfs_scmp(uint32_t a, uint32_t b) {
    return (int)(unsigned)(a - b);
}

// Convert from 32-bit little-endian to native order
static inline uint32_t lfs_fromle32(uint32_t a) {
    return (a);
}

// Convert to 32-bit little-endian from native order
static inline uint32_t lfs_tole32(uint32_t a) {
    return lfs_fromle32(a);
}
```