littlefs-project/littlefs #90: porting littlefs on STM32 with W25Q16JV Flash

e135193, 2018-08-15:
Hello,

      I would like to port the littlefs on my STM32 MCU with W25Q16JV flash. 

      In order to port it, What steps should I follow?

      Where should I put my SPI_Read and SPI_Write function in the littlefs source code?

Best Regards

fzhenyu, 2018-08-31:
Hi,
   I have done it on STM32f103.  the code shows below:
`int fs_flash_read(const struct lfs_config *cfg, lfs_block_t block,
        lfs_off_t off, void *buffer, lfs_size_t size)
{
    // Check if read is valid
    assert(off  % cfg->read_size == 0);
    assert(size % cfg->read_size == 0);
    assert(block < cfg->block_count);
    uint32_t addr = SPI_FLASH_FS_BASE_ADDR + block*4096 + off;
    int ret = lsspi_flash_read(buffer, addr, size);
    LSFS_DEBUG(DBG_LEVEL_DEBUG, "ls_flash_read ret=%d,[block,off,addr,size]=[%d, %d, 0x%08X, %d]\r\n", ret, block, off, addr, size);
//    for(int i = 0; i < 16; i ++)
//    {
//      if (i >= size)
//        break;
//      printf("%02X ", ((unsigned char *)buffer)[i]);
//    }
//    printf("\r\n");
    if (ret == -1)
    {
        return -1;
    }
    else
    {
        return 0;
    }
}

int fs_flash_prog(const struct lfs_config *cfg, lfs_block_t block,
            lfs_off_t off, const void *buffer, lfs_size_t size)
{
    // Check if write is valid
    assert(off  % cfg->prog_size == 0);
    assert(size % cfg->prog_size == 0);
    assert(block < cfg->block_count);  
    //printf("write block off size: %d-%d-%d\r\n", block, off, size);
    //uint32_t addr = block*4096 + off;
    uint32_t addr = SPI_FLASH_FS_BASE_ADDR + block*4096 + off;
    int ret = lsspi_flash_write((u8*)buffer, addr, size);
    LSFS_DEBUG(DBG_LEVEL_DEBUG, "ls_flash_write ret=%d [block,off,addr,size]=[%d, %d, 0x%08X, %d]\r\n", ret, block, off, addr, size);
    
//    for(int i = 0; i < 16; i ++)
//    {
//      if (i >= size)
//        break;
//      printf("%02X ", ((unsigned char *)buffer)[i]);
//    }
//    printf("\r\n");
    
    if (ret == -1)
    {
        return -1;
    }
    else
    {
        return 0;
    }
}

int fs_flash_erase(const struct lfs_config *cfg, lfs_block_t block)
{
    assert(block < cfg->block_count);  
    
    int block_addr = SPI_FLASH_FS_BLOCK_OFF + block;
    
    int ret = lsspi_flash_erase_sector(block_addr);
    LSFS_DEBUG(DBG_LEVEL_DEBUG, "fs_flash_erase ret=%d,[block, block_addr]=[%d, %d]\r\n",ret, block, block_addr);
    
    if (ret == -1)
    {
        return -1;
    }
    else
    {
        return 0;
    }
}

int fs_flash_sync(const struct lfs_config *c)
{
    return 0;
}`

fzhenyu, 2018-08-31:
`int fs_flash_read(const struct lfs_config *cfg, lfs_block_t block,
        lfs_off_t off, void *buffer, lfs_size_t size)
{
    // Check if read is valid
    assert(off  % cfg->read_size == 0);
    assert(size % cfg->read_size == 0);
    assert(block < cfg->block_count);
    uint32_t addr = SPI_FLASH_FS_BASE_ADDR + block*4096 + off;
    int ret = lsspi_flash_read(buffer, addr, size);
    LSFS_DEBUG(DBG_LEVEL_DEBUG, "ls_flash_read ret=%d,[block,off,addr,size]=[%d, %d, 0x%08X, %d]\r\n", ret, block, off, addr, size);

    if (ret == -1)
    {
        return -1;
    }
    else
    {
        return 0;
    }
}

int fs_flash_prog(const struct lfs_config *cfg, lfs_block_t block,
            lfs_off_t off, const void *buffer, lfs_size_t size)
{
    // Check if write is valid
    assert(off  % cfg->prog_size == 0);
    assert(size % cfg->prog_size == 0);
    assert(block < cfg->block_count);  

    uint32_t addr = SPI_FLASH_FS_BASE_ADDR + block*4096 + off;
    int ret = lsspi_flash_write((u8*)buffer, addr, size);
    LSFS_DEBUG(DBG_LEVEL_DEBUG, "ls_flash_write ret=%d [block,off,addr,size]=[%d, %d, 0x%08X, %d]\r\n", ret, block, off, addr, size);

    if (ret == -1)
    {
        return -1;
    }
    else
    {
        return 0;
    }
}

int fs_flash_erase(const struct lfs_config *cfg, lfs_block_t block)
{
    assert(block < cfg->block_count);  
    
    int block_addr = SPI_FLASH_FS_BLOCK_OFF + block;
    
    int ret = lsspi_flash_erase_sector(block_addr);
    LSFS_DEBUG(DBG_LEVEL_DEBUG, "fs_flash_erase ret=%d,[block, block_addr]=[%d, %d]\r\n",ret, block, block_addr);
    
    if (ret == -1)
    {
        return -1;
    }
    `else`
    {
        return 0;
    }
}

int fs_flash_sync(const struct lfs_config *c)
{
    return 0;
}`

e135193, 2018-08-31:
Hello, I can not see functions to read via SPI hardware. 

HAL_SPI_Transmit();
HAL_SPI_Receive();

Could you please share the full source code to my email address; enkavak@hotmail.com

fzhenyu, 2018-09-13:
stm32 has supported the hal driver.

geky, 2018-09-19:
Hi @e135193, you'll need to provide the driver to communicate to the W25Q16JV over the STM32 SPI bus. This code is different for each SPI driver.

It looks like this may be provided by the "lsspi" functions in @fzhenyu's example, but I'm unfamiliar with the STM libaries.

In Mbed OS this is provided by the SPIFBlockdevice class. It's in C++ but may help as a starting point:
https://github.com/ARMmbed/spif-driver/blob/master/SPIFBlockDevice.cpp

manymix, 2019-01-04:
Could you please send the source code for block device driver of lettleFS on STM32 + W25 mabymix@yahoo.com.

manymix, 2019-01-04:
Could you please send the source code for block device driver based HAL-CUBE STM32 + W25 manymix@yahoo.com.

visakh-png, 2020-01-06:
@fzhenyu . You have mentioned 4096 (block*4096). Is that your Block size. 
By seeing this It seems you are writing to and reading from absolute addresses , is that right?.
can you provide your config structure for LFS.

joserp93, 2022-03-24:
I would like to use littlefs on my STM32 MCU with W25N01 flash.

  Could someone provide me or send me a code already made that serves as a reference?

simon88, 2024-09-09:
Hi @joserp93 I'm in the same situation as you, have you succesfully use littlefs with w25n01?