littlefs-project/littlefs #92: Issue opening two files

jcalvarez2, 2018-08-25:
Hello,

First of all, great piece of code, very easy to integrate, did not have to touch anything, just follow the instructions and that's it. Thank you!

Now I have an issue. I tried to open 2 files at the same time, but I get LFS_ERR_NOMEM when I attempt to open the second file after successfully opening the first file. I'm trying to implement file copy. The system has no malloc, so I'm using static allocation for the buffers.

Any advice would be much appreciated.

Regards,

Jose

geky, 2018-08-25:
Hi @jcalvarez2!

Each file needs its own buffer, so if you're using the file_buffer in the global config struct, you are limited to only opening one file at a time.

However, recently @dpgeorge (https://github.com/ARMmbed/littlefs/pull/58) added the option to provide a static buffer with a per-file config struct. You'll need to change `lfs_file_open` to `lfs_file_opencfg`:
``` c
lfs_file_t file;
uint8_t file_buffer[PROG_SIZE];
const struct lfs_file_config file_cfg = {
    .buffer = file_buffer,
};

int err = lfs_file_opencfg(&lfs, &file, "file.txt", LFS_O_WRONLY, &file_cfg);
```

Then you can provide a buffer for each file without malloc.

Let me know if that helps or you have any other questions

jcalvarez2, 2018-08-25:
Hi @geky,

Many thanks for your prompt reply, much appreciated. Your explanation makes perfect sense. I will try using a per-file buffer.

jcalvarez2, 2018-08-28:
Issue closed. Thank you.

closed 2018-08-28 by jcalvarez2