littlefs-project/littlefs #95: Overhead calculator

X-Ryl669, 2018-09-03:
Can you provide some formula to figure out the overhead used for the filesystem compared to the data stored ?

Something with page size, and block size as input and the Y * (number of file of size XXX) ?

Thanks

geky, 2018-09-19:
Hi @X-Ryl669, sorry for the late response, it's taken me a bit to catch up on open issues.

This is the current minimums:
- 2 blocks per superblock
- 2 blocks per directory
- 1 block per file

So if you had 10 files stored in root, that would take:
2 for the superblock + 2 for directories (root is a directory) + 10 for files = 14 blocks at minimum

This is useful for most uses of embedded filesystems when files are smaller than the block size.

The full formula that handles when directory blocks/file blocks overflow is a bit more complicated. The full details can be found in the [SPEC.md](https://github.com/ARMmbed/littlefs/blob/master/SPEC.md). But if you want a direct formula it would be something like this:

![math](https://latex.codecogs.com/png.latex?%5C%5C%20%5Ctext%7Bblocks%20per%20file%7D%20%3D%20%5Cleft%20%5Clceil%20%5Cfrac%7B%5Ctext%7Bfile%20size%7D%7D%7B%5Ctext%7Bblock%20size%7D%20-%208%7D%20%5Cright%20%5Crceil%20%5C%5C%20%5C%5C%20%5C%5C%20%5Ctext%7Bblocks%20per%20dir%7D%20%3D%202%5Ctimes%5Cleft%20%5Clceil%20%5Cfrac%7B%5Csum%5E%7B%5Ctext%7Beach%20file%7D%7D%20%28%2012%20&plus;%20%5Ctext%7Bfile%20name%7D%20%29%7D%7B%5Ctext%7Bblock%20size%7D%20-%2020%7D%20%5Cright%20%5Crceil%20%5C%5C%20%5C%5C%20%5C%5C%20%5Ctext%7Btotal%20blocks%7D%20%3D%202%20&plus;%20%5Csum%5E%7B%5Ctext%7Beach%20dir%7D%7D%5Ctext%7Bblocks%20per%20dir%7D%20&plus;%20%5Csum%5E%7B%5Ctext%7Beach%20file%7D%7D%5Ctext%7Bblocks%20per%20file%7D)

---

Currently this isn't great when there are very few blocks (for example internal flash). But works ok for most forms of external flash. Most notably, small files always take up a full block at minimum.

As a sidenote: I'm currently working on this as a part of the 2.0 work which will get rid of the 2-block superblock requirement and allow small files to be inlined in their parent's directory blocks: https://github.com/ARMmbed/littlefs/pull/85

X-Ryl669, 2018-09-20:
Thanks a lot!

perigoso, 2021-12-16:
@geky for the v2 could you show what formulas one could use to calculate overhead?

kyrreaa, 2023-10-18:
What about the variable skip block section of each block. This varies with block number and basically grows with size of file?

geky, 2023-10-18:
@kyrreaa, great question! It's super unintuitive, but because the number of variable pointers forms a perfectly balanced binary tree, on _average_ the overhead never exceeds $2w$.

Since our word size $w$ is 32-bits, or 4 bytes, this is where the 8 comes from in the $\mathtt{file\\_blocks}$ equation:

```math
\mathtt{file\_blocks} = \left\lceil\frac{\mathtt{file\_size}}{\mathtt{block\_size}-8}\right\rceil
```

---

We can work through a couple examples to see this is true:

```
   0        1        2        3        4        5        6        7        8
.-----.  .-----.  .-----.  .-----.  .-----.  .-----.  .-----.  .-----.  .-----.
|     |<---ptr |<---ptr |<---ptr |<---ptr |<---ptr |<---ptr |<---ptr |<---ptr |
|     |<-|     |----ptr |<-|     |----ptr |<-|     |----ptr |<-|     |----ptr |
|     |<-|     |--|     |--|     |----ptr |<-|     |--|     |--|     |----ptr |
|     |<-|     |--|     |--|     |--|     |--|     |--|     |--|     |----ptr |
'-----'  '-----'  '-----'  '-----'  '-----'  '-----'  '-----'  '-----'  '-----'
 0 ptrs   1 ptrs   2 ptrs   1 ptrs   3 ptrs   1 ptrs   2 ptrs   1 ptrs   4 ptrs
'-----------.-----------'
     3 ptrs in 3 blocks
          3 < 6 (3*2)
'--------------------.--------------------'
              7 ptrs in 5 blocks
                   7 < 10 (5*2)
'--------------------------------------.--------------------------------------'
                               15 ptrs in 9 blocks
                                    15 < 18 (9*2)
```

One way to prove this is to look at each _row_ of pointers. Ignoring the first block, the first row has $n$ pointers, the second has $\frac{n}{2}$ pointers, the third has $\frac{n}{4}$, then $\frac{n}{8}$ etc:

```math
n + \frac{n}{2} + \frac{n}{4} +\frac{n}{8} + \cdots
```

This gives us a [geometric series](https://en.wikipedia.org/wiki/Geometric_series) that converges to $2n$, or an average of $2$ per block:

```math
n\cdot\sum_{i=1}^{\infty }{\left(\frac{1}{2}\right)^i} = n \cdot 2
```

kyrreaa, 2023-10-18:
Yeah I figured the number of pointers based on the block number outwards counting trailing zeroes in block number and add one. Since it is a bit convoluted the cost of the "actual size on disk" stat would be too high I guess.
Reason I ask is spending a few days hunting down why my superblocks disappeared. The problem went away when I started using a proper sized lookahead after finding your comment on a previous issue that a lookahead much bigger than the number of blocks in the fs may break something. Seems you were right. (I was writing 20-50 MB files till I got no more space error, but not all files were in memory when listed. Upon reboot filesystem was unmountable and superblocks were gone. My wish was to see if stopping before it was actually full would help.)

geky, 2024-02-28:
Hi @shahawi-sumup, those 28-bytes come from the CTZ skip-list overhead:

```math
\mathtt{file\_blocks} = \left\lceil\frac{\mathtt{file\_size}}{\mathtt{block\_size}-8}\right\rceil
```

littlefs carves out 8 bytes for each block for its own bookkeeping (the details are a bit more complicated, see [above](https://github.com/littlefs-project/littlefs/issues/95#issuecomment-1769071623)).

That being said, littlefs doesn't reserve a block for each file-metadata, the file's metadata is stored inside the parent directory. So you shouldn't need the `- 1*4KB (overhead (file_metadata))`.

Two things come to mind:

1. littlefs is trying to create copies of the metadata blocks to wear-level.

   Consider setting `block_cycles=-1` to disable block-level wear-leveling. At 16-blocks this may not get you much. littlefs still wear-levels metadata blocks internally, and data blocks are still dynamically wear-leveled via copy-on-write.

2. The extra blocks are needed for copies of the file's tail block when appending.

   It's rare for files to be perfectly block-size aligned, so sometimes littlefs needs to create a copy of the last block in a file in order to perform a write.

In general I would avoid trying to fill up littlefs completely. The copy-on-write system is kind of complex and can result in ENOSPC in strange situations like what you're seeing. I would save at least ~10% of storage for copy-on-write operations. Though in your case, at 16 blocks, saving 2 blocks (or maybe even just 1 if `block_cycles=-1`?) is probably sufficient.

geky, 2024-04-29:
This is being referenced externally, and a bit incorrectly (unfortunately the exact constraints are complicated), so I just wanted to clarify:

**littlefs _can_ fit in 2 blocks**, as of v2.

**Iff** all files are "inlineable", which requires:
1. they fit in RAM (`cache_size` specifically)
2. they fit in 1/4 block_size

Inline files are more expensive, using ~4x the storage, but this is better than storing small files in full blocks.

If you have a small amount of storage, say, 4 blocks, you may also want to consider using a larger block size, `block_size=2*block_size`, to allow for larger inline files and better storage utilization.