littlefs-project/littlefs #98: lfs_deorphan may enter infinity loop.

kneko715, 2018-09-13:
After some unexpected hardware reset, lfs_deorphan() entered an infinite loop and not return forever.
In that time, cwd.pair / cwd.head / cwd.d.tail took following value.

cwd.pair[0]=2;   cwd.pair[1]=3;
cwd.head[0]=0;  cwd.head[1]=0;
cwd.d.tail[0]=2;  cwd.d.tail[1]=3;

I know root cause is unexpected hardware reset, but lfs_deorphan() should detect infinity loop and return error code for caller.

Thank you.

geky, 2018-09-19:
Hi @kneko715 thanks for opening an issue. You're right we should probably make that loop terminate.

Although, littlefs should never really end up that state. What do you mean by hardware reset? Even if power is lost in the middle of a filesystem operation littlefs should be able to recover.

If you still have the broken disk image around, could you send me a copy so I can see if I can figure out that cause?

geky, 2018-09-19:
Patch here to break out of the inifinite loop https://github.com/ARMmbed/littlefs/pull/103

Thinking about possible failure cases, it's possible this could have happened if you interrupt lfs_format with a power failure. Do you think that might have been the source of the problem?

FreddieChopin, 2018-09-20:
Just a sidenote about the fix in #103 - the fixed behaviour won't be far from "infinite loop" on a very large device (several GB of SD card).

kneko715, 2018-09-21:
@getky thanks for reply and creating patch.

Sorry, broken disk image has already lost.

"unexpected hardware reset" means unexpected power failure.
At  that time, I was testing littlefs api.
The test case includes lfs_mount, lfs_format and lfs_unmount, so it's possibility that the power failure interrupted lfs_format.
 
The patch effective for our device. Thank you.

geky, 2018-09-21:
@FreddieChopin, good point. It seems like at least > 4 hours on a 4 GB SD card...

It's the best solution I can think for the short term. At least in [2.0](https://github.com/ARMmbed/littlefs/pull/85), format will only need to create 1 metadata pair, so it should be no longer possible to create this infinite loop during format.

closed 2018-09-27 by kneko715