littlefs-project/littlefs #105: CC Warning

snikeguo, 2018-09-22:
porting success for w25q128.
void lfs_init(void)
{
  cfg.read=lfs_read;
  cfg.prog=lfs_prog;
  cfg.erase=lfs_erase;
  cfg.sync=lfs_sync;
  cfg.read_size=4096;
  cfg.read_buffer=lfs_read_buffer;
  cfg.prog_size=4096;
  cfg.prog_buffer=lfs_prog_buffer;
  cfg.block_size=4096;
  cfg.block_count=100;
  cfg.lookahead_buffer=lfs_lookahead_buffer;
  cfg.lookahead=4096;
}
but cc warning.
[image]

snikeguo, 2018-09-22:
546: transfer of control bypasses initialization of:
Example:
int main(void){
   int choice = 1;
   int z =1;
   switch(choice)
   {
     case 1:
       int y = 1;
       z = y + z;
       break;
     case 2:
     break;
   }
   return 0;
In this example, y is an initialized variable that is in scope but is unused in the other cases. A transfer from the condition of the switch statement to a case label, bypassing the initialization of y, conflicts with the C++ Standard.
The usual way to fix this is to enclose the case that declares y in braces. The following code limits the scope of y to case 1, so an attempt to use it elsewhere causes an error:
case 1:   {
  int y = 1;
  z = y + z;
}
break;
Because y is a Plain Old Data (POD) type, an alternative is to not use initialization:
case 1:
 int y;
 y = 1;
 z = y + z;
 break;
This approach has the disadvantage that if code outside of case 1 uses y, and accidentally expects it to have the value assigned in case 1, no warning is given.

geky, 2018-09-26:
Hi @snikeguo, thanks for raising an issue

We're using different compilers with different flags so I haven't seen those warnings before. Added over here: https://github.com/ARMmbed/littlefs/pull/107

closed 2018-09-26 by geky