littlefs-project/littlefs #108: Get list of free blocks to erase them

bobatsar, 2018-09-26:
Hi there,

I am using a Serial NOR chip.

I can see in the traces, if a file grows over block size, a new block is first erased before it can be written to.

For performance reasons I want to reduce the expensive block_erase function on the flash chip (about 100ms for 4kb) as much as possible.

For this I test in the erase() function if a block is already erased (read of whole block and check if everything is 0xff) and return immediately.

This speeds up writing a file immensely if the flash memory was properly erased before.

To ensure the unused blocks can be erased e.g. in situations where the flash is not needed, I would need a traverse function to iterate over the next free blocks which would be taken for allocation, to ensure they are correctly erased.

What would be the best way to accomplish this?

geky, 2018-09-26:
Hi @bobatsar, quick answer:

The [lfs_alloc](https://github.com/ARMmbed/littlefs/blob/master/lfs.c#L287) function is responsible for allocating blocks. It keeps track of the variable lfs->free.i for iterating over a RAM backed bitmap called the "lookahead buffer". When [lfs->free.i >= lfs->free.size](https://github.com/ARMmbed/littlefs/blob/master/lfs.c#L289), the allocator scans for another lookahead buffer.

You could create a separate function to iterate over the lookahead buffer from lfs->free.i to lfs->free.size without modifying lfs->free.i. That would let you zero at most lfs->cfg->lookahead number of blocks.

It may require a bit of effort, but hopefully the existing [lfs_alloc](https://github.com/ARMmbed/littlefs/blob/master/lfs.c#L287) function would be a good starting point.

Out of curiosity, is this erasing being performed in a background thread?

bobatsar, 2018-09-27:
Hi @geky,

Thanks for the pointer to the lfs_alloc function.
I'll try to understand the lfs->free datastructure.

What is lfs->free.ack and lfs->free.off for?

If I have 4096 blocks and set the lookahead buffer also to 4096, it should span the entire memory.
Does this make sense or is it bad for performance? It would need 512 bytes for the lookahead buffer, which should be ok.

The erasing is planned to be done in idle phases, e.g. before going to a sleep mode.
We have quite strict timing requirements to write data every 10ms so the normal way of erasing of blocks is not working for us.

bobatsar, 2018-10-25:
Based on your tips I implemented it and I can "predelete" blocks in idle phases.

closed 2018-10-25 by bobatsar

harteware07, 2018-11-14:
@bobatsar 
that sounds very interesting, i too need this feature, i have to write large chunks of logging data from time to time, and could predelete in my idle task.
would you mind sharing your code with us?

bobatsar, 2018-11-14:
@harteware07  @geky 

Here is my implementation of the "predeletion" mechanism. I have no time for a PR and I am not sure if it can/should be integrated directly in LFS but I hope it helps you implementing it.

First some helper functions and the erased_blocks bitmap
```c
uint32_t erased_blocks[LFS_BLOCK_COUNT/32] = {0};
static bool _block_erase_finished = false;

bool block_bit_set(uint32_t *block_bitfield, lfs_block_t block){
    return block_bitfield[block/32] & (1U << (block % 32));
}

void set_block_bit(uint32_t *block_bitfield, lfs_block_t block){
    block_bitfield[block/32] |= (1U << (block % 32));
}

void clear_block_bit(uint32_t *block_bitfield, lfs_block_t block){
    block_bitfield[block/32] &= ~(1U << (block % 32));
}
```

In lfs_prog the written block has to be marked dirty.
```c
int lfs_prog(const struct lfs_config *c, lfs_block_t block,
                        lfs_off_t off, const void *buffer, lfs_size_t size)
    ...
    clear_block_bit(erased_blocks, block);
)
```

`is_block_empty()` reads a block and checks if it is empty. This is way faster than erasing it always.
So lfs_erase only erases blocks which are not yet predeleted.

```c
int lfs_erase(const struct lfs_config *c, lfs_block_t block){

    LOG_DEBUG("Erase block %i", block);
    if (block_bit_set(erased_blocks, block)){
        LOG_DEBUG("block %i seems to be already erased", block);
        return 0;
    }

    if (!is_block_empty(block)){
        LOG_DEBUG("erase 4kb block %i", block);
        erase_block(block*LFS_BLOCK_SIZE);
    }
    
    set_block_bit(erased_blocks, block);
    return 0;
}
```

And now the functions for periodically predeletion of blocks.

```c
static int lfs_traverse_create_used_blocks(void *p, lfs_block_t block) {
    uint32_t *used_blocks = p;
    set_block_bit(used_blocks, block);
    LOG_DEBUG("Block %i is in use", block);
    return 0;
}

/**
erase up to num_blocks of not yet erased but free blocks
timeout 0 means to erase all blocks
**/
int lfs_erase_blocks(lfs_t *lfs, int timeout_ms){
    uint32_t used_blocks[LFS_BLOCK_COUNT/32] = {0};

    // create free list
    lfs_check_error(lfs_traverse(lfs, lfs_traverse_create_used_blocks,
                                 &used_blocks));

    int empty_blocks = 0;
    bool blocks_erased = false;
    uint64_t start_time = sysclock_ms();
    // find next block starting from free.i
    for (int i=0; i<LFS_BLOCK_COUNT; i++){
        if (timeout_ms && sysclock_ms() >= start_time + timeout_ms){
            break;
        }

        lfs_block_t block = (lfs->free.i + i) % LFS_BLOCK_COUNT;

        // take next free marked block
        if (!block_bit_set(used_blocks, block)){
            // check if it is empty
            if (!block_bit_set(erased_blocks, block)){
                // if the block is marked as free but not yet marked as erased,
                // erase it
                LOG_DEBUG("Block %i is free but not yet marked erased",
                                                                        block);
                // erase if needed
                lfs_erase(lfs->cfg, block);
                blocks_erased = true;
            }
            empty_blocks += 1;
        }
    }

    if (!blocks_erased){
        _block_erase_finished = true;
    }

    LOG_INFO("%i empty_blocks starting from %i available in %ims",
                 empty_blocks, lfs->free.i, sysclock_ms()-start_time);
    return empty_blocks;
}
```

`_block_erase_finished` is reset at e.g. startup and the Idle task calls 
`lfs_erase_blocks(100)` until `_block_erase_finished` is set

If you find some bugs or improvements, please tell me. Feel free to use it. Perhaps it ends up in a PR?

harteware07, 2018-11-14:
@bobatsar 
thank you for your quick reply! I will test it as soon i have lfs working on my system! have you had the chance to measure maximal throughput?

ajaybhargav, 2019-02-16:
@bobatsar Thanks for sharing your implementation, How is the write performance after the pre-delete implementation?