littlefs-project/littlefs #110: lfs_rename does not always work

ohoc, 2018-10-01:
Hi all,
I am having problems with lfs_rename(): in certain cases, the file is not renamed, and I have traced it back to when lfs_dir_append() needs to allocate a new block.

I have attached a small test program below that reproduces the problem.
I would expect to have only ".txt" files listed, but each 7th file is not renamed and is still a ".tmp".
The problem occurs more often if the block size is smaller, and less often when the block size is bigger.

Any idea where the problem can be?

Thanks.

```C
#include <string.h>
#include <stdio.h>

#include "lfs.h"

#define TOTAL_SIZE   (1024*1024)
#define BLOCK_SIZE   256
#define BLOCK_COUNT  (TOTAL_SIZE/BLOCK_SIZE)

int ram_read(const struct lfs_config *c, lfs_block_t block,
        lfs_off_t off, void *buffer, lfs_size_t size);
int ram_prog(const struct lfs_config *c, lfs_block_t block,
        lfs_off_t off, const void *buffer, lfs_size_t size);
int ram_erase(const struct lfs_config *c, lfs_block_t block);
int ram_sync(const struct lfs_config *c);

uint8_t ramfs[TOTAL_SIZE];
lfs_t lfs;
lfs_file_t file;

const struct lfs_config cfg = {
    .read  = ram_read,
    .prog  = ram_prog,
    .erase = ram_erase,
    .sync  = ram_sync,

    .read_size = 16,
    .prog_size = 16,
    .block_size = BLOCK_SIZE,
    .block_count = BLOCK_COUNT,
    .lookahead = 128,
};

int main(void) {

    int err;
    lfs_format(&lfs, &cfg);
    lfs_mount(&lfs, &cfg);

    char filename1[64];
    char filename2[64];
    for (unsigned int i = 0; i < 32; ++i) {
        sprintf(filename1, "testfile_%08X.tmp", i);
        sprintf(filename2, "testfile_%08X.txt", i);
        lfs_file_open(&lfs, &file, filename1, LFS_O_RDWR | LFS_O_CREAT);
        lfs_file_write(&lfs, &file, "asdf\n", 5);
        lfs_file_close(&lfs, &file);
        lfs_rename(&lfs, filename1, filename2);
    }

    lfs_dir_t dir;
    lfs_dir_open(&lfs, &dir, ".");

    struct lfs_info info;
    while (true) {
        int res = lfs_dir_read(&lfs, &dir, &info);
        if (res < 0) {
            break;
        }

        if (res == 0) {
            break;
        }

        switch (info.type) {
            case LFS_TYPE_REG: printf("reg "); break;
            case LFS_TYPE_DIR: printf("dir "); break;
            default:           printf("?   "); break;
        }

        static const char *prefixes[] = {"", "K", "M", "G"};
        for (int i = sizeof(prefixes)/sizeof(prefixes[0])-1; i >= 0; i--) {
            if (info.size >= (1 << 10*i)-1) {
                printf("%*u%sB ", 4-(i != 0), info.size >> 10*i, prefixes[i]);
                break;
            }
        }

        printf("%s\n", info.name);
    }

    lfs_dir_close(&lfs, &dir);

    lfs_unmount(&lfs);
}

int ram_read(const struct lfs_config *c, lfs_block_t block,
        lfs_off_t off, void *buffer, lfs_size_t size)
{
    memcpy(buffer, ramfs + block*c->block_size + off, size);
    return 0;
}

int ram_prog(const struct lfs_config *c, lfs_block_t block,
        lfs_off_t off, const void *buffer, lfs_size_t size)
{
    memcpy(ramfs + block*c->block_size + off, buffer, size);
    return 0;
}

int ram_erase(const struct lfs_config *c, lfs_block_t block)
{
    return 0;
}

int ram_sync(const struct lfs_config *c)
{
    return 0;
}
```

geky, 2018-10-07:
Hi @ohoc, thanks for the great bug report!

Sorry, about the delay, your example code made it easy to track down what was going wrong. Looks like it was an overlooked corner case when a rename causes a directory to split into a linked-list.

Should be fixed in https://github.com/ARMmbed/littlefs/pull/111. Let me know if I'm overlooking anything.

ohoc, 2018-10-12:
Hi @geky,
Sorry for the late reply.
Your fix seems to work well, thank you for your help!

closed 2018-10-20 by geky