littlefs-project/littlefs #115: Is there any requirement on deleted blocks (all 0xFFFF..., all 0x0000...)?

ProfDrYoMan, 2018-10-18:
Hi,

I can not find any clear information if it is a requirement that deleted blocks show data 0xFFFF...

I might use this FS for the case that deleted blocks show data 0x0000...

Is this ok with the current implementation? If not, what needs to be changed to get this done?

Thanks in advance.

geky, 2018-10-18:
Hi @ProfDrYoMan, I like your username. 

Deleted blocks showing 0x0000 is not an issue. littlefs doesn't trust blocks it hasn't erased itself and always erases before writing to a block it does not know about.

ProfDrYoMan, 2018-10-18:
Ok, nice to here that it still works if deleted flash blocks show 0X0000... (which is not very common). 

Thanks for the fast answer. Will put it into a test project tomorrow.

geky, 2018-10-18:
Out of curiousity, what type of storage are you using that erases to 0x0000?

ProfDrYoMan, 2018-10-22:
It's an Infineon AURIX.

closed 2018-10-29 by ProfDrYoMan