littlefs-project/littlefs #116: Performance issue with large file header update

mstempin, 2018-10-22:
I use LFS to write onto a 512Mb SPI Flash device featuring 256kb sector/block erase size and 512 byte program page size.

I need to write a 4MB file, preceded by a short header of 8 bytes (basically, its applicative length and checksum), that I can update until the whole file is written:

- seek 8 bytes from the beginning
- write 4 MB
- seek to beginning
- write 8 bytes

After this last step, it takes forever to flush the file, as lfs_file_flush is performing a byte copy over the whole 4MB...

conkerkh, 2018-10-23:
I discussed this some time ago, have a look #104 I ended up with slightly modified code from what sub-op posted but this approach works in general. If you are not implementing Ring Buffer and header is for something else simply keep header in separate file.

bobatsar, 2018-10-25:
Have a look at https://github.com/ARMmbed/littlefs/blob/master/DESIGN.md#ctz-skip-lists 
Append works fast but changing a file in the beginning makes all blocks afterwards to be changed.

mstempin, 2018-10-25:
Yes, I know, but in this case it is not just changing all the blocks afterwards, it does a bytewise copy of the whole 4MB file, which is far from optimal.

In my case I had the opportunity to change my design to move the file header as a trailer at the end of the file instead but this may be a problem in some other situations.