littlefs-project/littlefs #117: Can I open multiple files at the same time

Wanlq, 2018-10-23:
I want opent multiple files at the same time but the code can't run with my mind
my code is below
//lfs 
#define LFS_TEST_DEMO
#ifdef LFS_TEST_DEMO
lfs_t g_lfs;
lfs_file_t g_lfs_file[2];
struct lfs_config	g_lfs_cfg;

uint8_t	lfs_read_buf[256] = {0};
uint8_t	lfs_prog_buf[256] = {0};
uint8_t	lfs_lookahead_buf[256] = {0};
uint8_t	lfs_file_buf[256] = {0};

static int lfs_fuse_statfs_count(void* p,lfs_block_t block)
{
	*(lfs_size_t*)p += 1;
	return 0;
}

void lfs_free_spcae(u32* free_space)
{
	lfs_size_t use_space = 0;
	int err = lfs_traverse(&g_lfs,lfs_fuse_statfs_count,&use_space);
	if(err)
	{
		return ;
	}
	*free_space = g_lfs_cfg.block_count - use_space;
}

void lfs_set_config(void)
{
	// configuration of the filesystem is provided by this struct
	g_lfs_cfg.read  = block_device_read,
	g_lfs_cfg.prog  = block_device_prog,
	g_lfs_cfg.erase = block_device_erase,
	g_lfs_cfg.sync  = block_device_sync,

	
	g_lfs_cfg.read_size = 256,
	g_lfs_cfg.prog_size = 256,
	g_lfs_cfg.block_size = 4096,
	g_lfs_cfg.block_count = 256,
	g_lfs_cfg.lookahead = 256,

	g_lfs_cfg.read_buffer = lfs_read_buf;
	g_lfs_cfg.prog_buffer =	lfs_prog_buf;
	g_lfs_cfg.lookahead_buffer = lfs_lookahead_buf;
	g_lfs_cfg.file_buffer = lfs_file_buf;	
}

void lfs_test_demo(void)
{
	u8 lfs_write_buf1[10] = {"123456789"};
	u8 lfs_write_buf2[10] = {"abcdefghi"};
	u8 lfs_read_buf1[10] =  {0};
	u8 lfs_read_buf2[10] =  {0};
	lfs_set_config();

    // mount the filesystem
    int err = lfs_mount(&g_lfs, &g_lfs_cfg);

    // reformat if we can't mount the filesystem
    // this should only happen on the first boot
    if (err) {
    	LFS_DEBUG("lfs need formart flash to mount filesystem: %d\n", __LINE__);
        lfs_format(&g_lfs, &g_lfs_cfg);
        lfs_mount(&g_lfs, &g_lfs_cfg);
    }

#if 0
    err = lfs_file_rewind(&g_lfs, &g_lfs_file1);
    if (err) {
    	LFS_DEBUG("lfs rewind error: %d\n", __LINE__);
    }   
 
    err = lfs_file_rewind(&g_lfs, &g_lfs_file2);
    if (err) {
    	LFS_DEBUG("lfs rewind2 error: %d\n", __LINE__);
    }
#endif
	//make dir
	lfs_mkdir(&g_lfs, "E:\\");

    // read current count
    err = lfs_file_open(&g_lfs, &g_lfs_file[0], "boot_count1.txt", LFS_O_RDWR | LFS_O_CREAT);
    if (err) {
    	LFS_DEBUG("lfs open1 error: %d\n", __LINE__);
    }    

    err = lfs_file_open(&g_lfs, &g_lfs_file[1], "boot_count2.txt", LFS_O_RDWR | LFS_O_CREAT);
    if (err) {
    	LFS_DEBUG("lfs open2 error:%d  LINE %d\n",err, __LINE__);
    }    

    err = lfs_file_write(&g_lfs, &g_lfs_file[0], lfs_write_buf1, sizeof(lfs_write_buf1));
    if (err) {
    	LFS_DEBUG("lfs write1 error: %d\n", __LINE__);
    }

    err = lfs_file_write(&g_lfs, &g_lfs_file[1], lfs_write_buf2, sizeof(lfs_write_buf2));
    if (err) {
    	LFS_DEBUG("lfs write2 error: %d\n", __LINE__);
    }

	//if not set cursor will read data error
	lfs_file_seek(&g_lfs, &g_lfs_file[0],0,LFS_SEEK_SET);
    lfs_file_seek(&g_lfs, &g_lfs_file[1],0,LFS_SEEK_SET);

    
    err = lfs_file_read(&g_lfs, &g_lfs_file[0], lfs_read_buf1, sizeof(lfs_read_buf1));
    if (err) {
    	LFS_DEBUG("lfs read error: %d\n", __LINE__);
    }   

    err = lfs_file_read(&g_lfs, &g_lfs_file[1], lfs_read_buf2, sizeof(lfs_read_buf2));
    if (err) {
    	LFS_DEBUG("lfs read error: %d\n", __LINE__);
    }  
	
  	LFS_DEBUG("lfs read1 error: %s %d\n",lfs_read_buf1, __LINE__);
  	LFS_DEBUG("lfs read2 error: %s %d\n",lfs_read_buf2, __LINE__);
    // remember the storage is not updated until the file is closed successfully
    err = lfs_file_close(&g_lfs, &g_lfs_file[0]);
    if (err) {
    	LFS_DEBUG("lfs close1 error: %d\n", __LINE__);
    } 

    err = lfs_file_close(&g_lfs, &g_lfs_file[1]);
    if (err) {
    	LFS_DEBUG("lfs close2 error: %d\n", __LINE__);
    } 
    
    // release any resources we were using
    err = lfs_unmount(&g_lfs);
    if (err) {
    	LFS_DEBUG("lfs ummount error: %d\n", __LINE__);
    } 
    
    // print the boot count
    LFS_DEBUG("boot_count: %s\n", lfs_read_buf1);
	LFS_DEBUG("boot_count: %s\n", lfs_read_buf2);
}
#endif

Wanlq, 2018-10-23:
I can open multiple files at the same time if I use the malloc function, but I cannot do this with a static array
//code 
#if 0
	g_lfs_cfg.read_buffer = lfs_read_buf;
	g_lfs_cfg.prog_buffer =	lfs_prog_buf;
	g_lfs_cfg.lookahead_buffer = lfs_lookahead_buf;
	g_lfs_cfg.file_buffer = lfs_file_buf;	
#else
	g_lfs_cfg.read_buffer = NULL;
	g_lfs_cfg.prog_buffer =	NULL;
	g_lfs_cfg.lookahead_buffer = NULL;
	g_lfs_cfg.file_buffer = NULL;
#endif
//use malloc funtion to get mem

geky, 2019-01-28:
Hi @Wanlq, sorry for the very late response.

@dpgeorge actually fixed this in https://github.com/ARMmbed/littlefs/pull/58. But the "cfg.file_buffer" variable still exists for backwards compatibility.

If you use the `lfs_file_opencfg` function you can provide a buffer for each file:
``` c
lfs_t g_lfs;
lfs_file_t g_lfs_file[2];
uint8_t	lfs_file_buf[2][256] = {{0}, {0}};
const struct lfs_file_config g_lfs_file_cfg[2] = {
    { // config for file[0]
        .buffer = &lfs_file_buf[0];
    },
    { // config for file[1]
        .buffer = &lfs_file_buf[1];
    }
};

// read current count
err = lfs_file_opencfg(&g_lfs, &g_lfs_file[0], "boot_count1.txt",
        LFS_O_RDWR | LFS_O_CREAT, &g_lfs_file_cfg[0]);
if (err) {
    LFS_DEBUG("lfs open1 error: %d\n", __LINE__);
} 
```