littlefs-project/littlefs #122: LittleFS setup on NOR flash

ste-camp, 2018-10-28:
Hello, I need some help setting up LittleFS.
I'm programming a custom OS for embedded devices and I'm trying to integrate little fs in it for flash filesystem management, but somehow I can't get to make it work properly and documentation is not very detailed on the meaning of the config parameters:
I have an SPI NOR Flash (this one [Winbond Electronics W25Q64FVSSIG](https://www.arrow.com/en/products/w25q64fvssig/winbond-electronics) ) and this is my config code:

```c
const struct lfs_config cfg = {
        // block device operations
        .read  = lfs_read_wrapper,
        .prog  = lfs_write_wrapper,
        .erase = lfs_erase_wrapper,
        .sync  = lfs_sync_wrapper,

        // block device configuration
        .read_size = (256),
        .prog_size = (256),
        .block_size = (4096),
        .block_count = 2048,
        .lookahead = 256,
};
```
The error I'm getting is that even after I completely erase the flash chip and call lfs_format() I get LFS_ERR_NOSPC error, but makes no sense because the chip has just been deleted.

Am I doing something wrong? Or maybe lfs doesn't support this kind of flash?

Thank you

FabianInostroza, 2019-01-13:
I had a similar issue, I solved it by adding some delay after/before the CS toggling.

michaelkollmann, 2019-01-22:
> 
> 
> I had a similar issue, I solved it by adding some delay after/before the CS toggling.

After / before the CS toggling. What's a "CS toggling"?

FabianInostroza, 2019-01-22:
@michaelkollmann The assertion/deassertion of the flash CS (or chip enable) pin between SPI transfers.

For the record, I faced a similar issue on a STM32L0 using libopencm3, the cause was bug in the silicon (it corrupted the last bit in the data register), see section 2.6.3 of the STM32L058 errata.

michaelkollmann, 2019-01-23:
Thanks! I'm pretty new to the whole embedded development world, that's why I was confused by the CS thingy.

The solution to the problem I was having was, that the SPI driver on my cpu only allowed a maximum data length of 255B. In the littlefs config I set the page sitze to 256B. This resulted in an invalid cast which caused the read and write to fail.

All I had to do was to separate each read and write call into two separate calls to my memory, offset by 128B.