littlefs-project/littlefs #124: reentrency (read file while write)

ranshalit, 2018-11-01:
Hello,

I am using littlefs with freertos (nxp sdk).
Is it allowed to read file while another task is writing into the same file ?

Thank you,
ranran

cfr-mir, 2019-01-02:
Ranran,

I am using littleFS with FreeRTOS.

My experience are, Multible readers are OK, but writers must be alone.

If reads are occuring while a file are open for write, you will have bad data written into your file.

Regards
Carsten

geky, 2019-01-25:
Hi, sorry for late response.

littlefs has no locking of its own (this makes porting easier), so if you use littlefs in multiple threads you will need to lock each call with your own mutex.

But if you add locks both multiple readers and writers should work.

The fact that multiple lockless readers works is probably luck due to the fact that reads don't change much state. I'm not sure it will work if files span multiple blocks.

cfr-mir, 2019-01-25:
Geky and Ranran,
Sorry for not beeing super precise. Of course you must have a mutex protection around all calls to LittleFS. I have that.
In the meantime I have investigated my concurrent write/read problem. And It originates from newlibs filestream implementation, which according to specs should be reentrant (With the proper mutexes implemented).

So LittleFS do support multiple open files reading and writing, as long as you keep a mutex around calls to LittleFS. Geky you knew that.

Best regards
Carsten