littlefs-project/littlefs #130: LittleFS writes too many bytes when data size more than program size

TheRedShed, 2018-12-06:
I am using LittleFS with an SPI flash chip. I have my read and program sizes set to 32 bytes with SPI buffers of 40 bytes (for SPI commands preceding the data bytes). When I attempt to write more bytes than that set for my program size, LittleFS attempts to write too many bytes, overflowing my SPI receive bufer, and crashing the system.

I have traced the offending code to be this line
https://github.com/ARMmbed/littlefs/blob/ec4d8b68add6a7de021dc09ef08123ab323cbc38/lfs.c#L188

What happens is if I need to write, say, 88 bytes, then this line of code returns 64 which is then sent to the program function
https://github.com/ARMmbed/littlefs/blob/ec4d8b68add6a7de021dc09ef08123ab323cbc38/lfs.c#L189
which overflows my SPI transmit and receive buffers.

I made the following modification:
```
lfs_size_t diff = size - (size % lfs->cfg->prog_size);
if (diff > lfs->cfg->prog_size)
{
    diff = lfs->cfg->prog_size;
}
int err = lfs->cfg->prog(lfs->cfg, block, off, data, diff);
```
With this fix, the number of bytes written does not exceed the set program size and my application no longer crashes.

Ultimately, my question is: Is this a valid fix? I don't have a full understanding of the internal workings of LittleFS and am not sure of the impact of this fix. Yes, I could just bump up my program size, but my limit is 256 bytes per SPI write for this flash chip and the problem remains if I try to write more than that.

FreddieChopin, 2018-12-06:
> Ultimately, my question is: Is this a valid fix?

It is not. It should be perfectly legal to write/read/erase multiples of minimal write/read/erase blocks and if your driver does not support that you should either fix your driver or handle that in the read/write/erase wrappers (just use a loop which reads/writes/erases smaller blocks).

TheRedShed, 2018-12-07:
Oh, I see. Okay. Thank you for the response.

closed 2018-12-07 by TheRedShed