littlefs-project/littlefs #138: Several directory edge cases cause FS corruption

TheLoneWolfling, 2019-01-28:
There are several directory edge cases that currently (sometimes) corrupt the filesystem, mainly due to uninitialized variables

Some I have found so far:

lfs_remove(&lfs, "");
lfs_remove(&lfs, ".");
lfs_remove(&lfs, "..");
lfs_remove(&lfs, "/");
lfs_remove(&lfs, "//"); // Or indeed any number of "/"s
Any of the above strings with mkdir.
Any of the above strings with rename.

geky, 2019-02-12:
Hi @TheLoneWolfling, sorry for the delay, this is a really good catch. It looks like there was an issue with expectations around what lfs_dir_find would populate when passed the root directory. I'll push up a fix on the v2-alpha in a bit.

Thanks for posting the alternative dir traversal function! You're very much right '..' is damn tricky. I think the path parsing is up there with the most complicated parts of littlefs. :)

I actually use unbounded lookahead in the dir parsing logic ([here](https://github.com/ARMmbed/littlefs/blob/v2-alpha/lfs.c#L1094)). This tradeoff is a part of sticking with unbounded RAM. You can actually use file paths of any length without increasing the RAM consumption (though storage usage may increase).

And while the dir parsing is O(n^2), it doesn't need to touch the disk during these operations. So in reality it's not noticeable. (and there's usually not _that_ many '..'s in a path).

I do think littlefs is currently parsing paths correctly (here's the [test cases](https://github.com/ARMmbed/littlefs/blob/master/tests/test_paths.sh#L84), feel free to add more). Let me know if I'm wrong though.

TheLoneWolfling, 2019-02-13:
Ah, so _that's_ what that double-loop does.

Yep, this seems to be fixed with a06447. I'll open another ticket if I see anything else.

closed 2019-02-13 by TheLoneWolfling