littlefs-project/littlefs #140: What error codes are block device operations allowed to return?

TheLoneWolfling, 2019-01-28:
And when a block device operation returns a non-zero result, is that considered a fatal error?

(By "block device operation" I mean lfs_config.read/erase/prog/sync.)

TheLoneWolfling, 2019-01-28:
Related to #137.

FreddieChopin, 2019-01-28:
From my understanding everything you return which is not a `0` (or `LFS_ERR_OK`) is treated as an error. The exact value does not matter, it should be propagated to the user with no changes.

geky, 2019-01-28:
@FreddieChopin is correct.

Rules for all bd operations:
- `0` (or `LFS_ERR_OK`) indicates that a write is successful
- `LFS_ERR_CORRUPT` indicates that the block is corrupt. On erase/prog/sync littlefs will drop the block and try to find another block to use. On read littlefs will error if it needs the data on the block. Though note some reads are to "uninitialized" block, littlefs is fine with "uninitialized" blocks returning `LFS_ERR_CORRUPT` if they do so consistently. This makes it easier to take advantage of ECC on block devices, you can just return `LFS_ERR_CORRUPT` when the ECC is invalid.
- Negative values are treated as errors, littlefs will stop what it's doing and return the error to the user as is.
- Positive values - I've just realized these aren't handled well. If a block device operation returns a positive value littlefs will probably break. Some of the internal layers use the positive range of integers to pass return values. This isn't ideal but not very easy to prevent.

geky, 2019-01-30:
Yep 👍

After any error LittleFS should continue to work correctly without needing a remount.

There's a special note for lfs_mount/lfs_dir_open/lfs_file_open in that these functions will need to be repeated as errors will leave the underlying structs uninitialized.

If an error prevents a file write, that file will not be committed when it is closed. We decided this was the best approach since block device errors are common at the end of a block device's lifespan, you need to close a file to clean up any memory it is using, and allowing files to become corrupted at the end of a block device's lifespan sounded like a bad idea.

Though I will admit that there is currently variable amounts of testing on errors.

TheLoneWolfling, 2019-01-31:
In which case I have a few bug reports to write :)

Thank you!

closed 2019-01-31 by TheLoneWolfling