littlefs-project/littlefs #141: Assert in lfs_file_write following lfs_file_truncate on file larger than block size

ebinans, 2019-02-01:
I get the following if file is bigger than block size and it is truncated prior to writing:
littlefs/lfs.c:183: lfs_cache_prog: Assertion `pcache->block == 0xffffffff' failed.

See attached code reproducing the issue on v1.7.1
[riplfs.tar.gz](https://github.com/ARMmbed/littlefs/files/2822068/riplfs.tar.gz)

ebinans, 2019-04-18:
The issue is still present in v2.0.3.

The following helps however not knowing internals much I'm not 100% sure it is correct fix: [patch.zip](https://github.com/ARMmbed/littlefs/files/3093474/patch.zip)

geky, 2019-05-22:
Thanks for creating an issue and test case! Sorry it took me so long to get to it.

Your fix is correct, though it does mean we lose the chance to reuse the cache if we write to the block we truncate to. This can happen if the file is smaller than the block size.

Though the previous code was definitely not doing this correctly.

I've pushed up a different fix here: https://github.com/ARMmbed/littlefs/commit/8bf45c3a294a65392d39da738fa4b8775e79b605

Also this highlighted a big hole in the littlefs tests around truncation, so I added a couple more tests, should cover the bug you've found.

I'll merge these once CI is passing (currently seems that GitHub is having webhook issues) unless you notice any issues.

ebinans, 2019-05-30:
All looks good now, thank you!

closed 2019-05-30 by ebinans