littlefs-project/littlefs #144: lfs_unmount segfaults when lfs hasn't been mounted.

TheLoneWolfling, 2019-02-13:
Could you please check that `lfs->cfg` is non-`NULL` in `lfs_deinit`?

As-is, if you call `lfs_unmount` with a `struct lfs` that hasn't been passed to `lfs_mount`, it typicially segfaults, as `lfs->cfg` is generally `NULL`.

If this is an invalid case, please mention it in the header.

This is with the v2 alpha, commit c810747eb417ec263303b733e9cce4af2e37df89.

TheLoneWolfling, 2019-02-13:
Hmm. It's actually a little more widespread than this. 

I'd typically assume that calling `unmount` on a volume that's already unmounted is ok to do. Ditto, I'd typically assume closing an already-closed file is ok to do.

However, both of these end up with double-frees. (Of `lfs->rcache.buffer` / `lfs->pcache.buffer` / `lfs->free.buffer` / `file->cache.buffer`.) Perhaps set these to NULL after freeing them?

ajaybhargav, 2019-02-17:
faced the same issue, my current system does not free an already freed buffer which causes system crash. I kept a flag in a wrapper (taken from mbed os library) to keep track of mount status. and flag is checked in almost all operations before I actually call the lfs APIs in wrapper.