littlefs-project/littlefs #145: Inadvertantly reopening a file can hang LFS.

TheLoneWolfling, 2019-02-13:
The following code:

```
#include <assert.h>
#include <stdlib.h>
#include "lfs.h"
#include "emubd/lfs_emubd.h"

#define REQUIRE(x) do {if(!(x)) {assert(false); exit(1);}} while(0);

lfs_t lfs;
lfs_emubd_t bd;

const lfs_config cfg = {
    .context          = &bd,
    .read             = &lfs_emubd_read,
    .prog             = &lfs_emubd_prog,
    .erase            = &lfs_emubd_erase,
    .sync             = &lfs_emubd_sync,

    .read_size        = 128,
    .prog_size        = 128,
    .block_size       = 1024,
    .block_count      = 1024,
    .block_cycles     = 4096,
    .cache_size       = 128,
    .lookahead_size   = 512*1024,
    .read_buffer      = NULL,
    .prog_buffer      = NULL,
    .lookahead_buffer = NULL,
    .name_max         = 255,
    .file_max         = 16384,
    .attr_max         = 16,
};

int main(void) {
    lfs_emubd_create(&cfg, "blocks");

    REQUIRE(lfs_format(&lfs, &cfg) == 0);
    REQUIRE(lfs_mount(&lfs, &cfg) == 0);
    
    {
        lfs_file_t fileA;
        REQUIRE(lfs_file_open(&lfs, &fileA, "A", LFS_O_CREAT | LFS_O_RDONLY) == 0);
    }
    {
        lfs_file_t fileB;
        REQUIRE(lfs_file_open(&lfs, &fileB, "B", LFS_O_CREAT | LFS_O_RDONLY) == 0);
    }
};
```

Hangs forever in `lfs_dir_commit` in the second open.

Note: you need optimizations on for this hang to occur.

The proximal cause here is this:

```
static int lfs_dir_commit(lfs_t *lfs, lfs_mdir_t *dir,
        const struct lfs_mattr *attrs, int attrcount) {
    // check for any inline files that aren't RAM backed and
    // forcefully evict them, needed for filesystem consistency
    for (lfs_file_t *f = (lfs_file_t*)lfs->mlist; f; f = f->next) {
        if (...) {
            ...
        }
    }
```
If f == f->next when we hit here (and the if statement doesn't do something to kick us out of it)... we hang forever.

In the original example, the compiler is realizing that fileA and fileB are never in scope at the same time, so it's using the same stack address for both. As we didn't close the file, lfs still has a dangling reference in the form of lfs->mlist. Then when we add the second file to the list of files we point file->next _to itself_... which causes issues.

Hence why you need sufficient optimizations on: FileA and fileB must end up getting allocated to the same stack address.

(Alternatively, have them both use `fileA`, although that way's cheating :-) as it gives a big hint as to what's going on.)

I could easily see this occurring if e.g. someone forgets to close a file, it goes out of scope, and then something causes lfs to flush the directory. Hangs are never particularly fun to debug.

Found on commit c810747eb417ec263303b733e9cce4af2e37df89.

TheLoneWolfling, 2019-02-13:
Personally I suggest that lfs_file_open should take a zeroed struct, although note that this is a backwards-incompatible change.

That way you can e.g. fairly easily add an is_valid flag for debug.

apmorton, 2019-03-03:
It's an unfortunate and sharp corner, but in the end this really isn't a bug in littlefs per say, and there isn't really anything you can do in littlefs to reliability detect this kind of bug.

Adding an assert in the loop to make sure `file->next != file` would certainly make this specific error easier to spot, but it still doesn't do anything for the case where you open a file on the stack and then simply forget to close it before returning.

You will end up with a pointer in the middle of a linked list to invalid stack memory, which is just a time bomb waiting to detonate.

The most foolproof way to avoid this issue is to compile your code as c++ and wrap this part of the lfs API with some simple raii file handle types that enforce file closing in their destructor.

If that isn't an option for you, the best you can really do if you must allocate file structures on the stack is to add the assert to lfs and cross your fingers. The assert is definitely more effective if you do not allocate file structures on the stack - at the very least in that case you can reasonably assert that any clobbering would be done in lfs itself (ie, opening into a structure that is already "open"), not at some random time in the future when your stack location gets reused.

TheLoneWolfling, 2019-03-03:
> there isn't really anything you can do in littlefs to reliability detect this kind of bug.

Some suggestions:

1. Document that the lfs_file_t should be zeroed when passed to lfs_file_open.
2. Have lfs_file_open assert !is_valid (or something similar). (And set is_valid, and lfs_file_close assert is_valid and set is_valid false.).
3. Have operations that iterate over files assert is_valid on said files.
4. Have a counter num_open_files. Assert if it goes negative.
4. Have operations that iterate over files assert if they've looped "too many times" (num_open_files? Max number of files that theoretically can fit into the currently-open filesystem? Something like that.).

Is this perfect? No. Is this much better at catching this sort of thing that the current approach? Yes.

As-is, I am hesitant to recommend a filesystem that states it's a "A little fail-safe filesystem" and yet doesn't do basic sanity checks like the above.

apmorton, 2019-04-10:
Sorry for the late reply - life gets in the way.

>1. Document that the lfs_file_t should be zeroed when passed to lfs_file_open.
>2. Have lfs_file_open assert !is_valid (or something similar). (And set is_valid, and lfs_file_close assert is_valid and set is_valid false.).

As you mentioned, this would be a breaking change.
Maybe this check could be gated behind a preprocessor define to retain backwards compatibility.
In any case, this would not help identify the issue you reported.

```c
{
    lfs_file_t fileA = { 0 };
    REQUIRE(lfs_file_open(&lfs, &fileA, "A", LFS_O_CREAT | LFS_O_RDONLY) == 0);
}
{
    lfs_file_t fileB = { 0 };
    REQUIRE(lfs_file_open(&lfs, &fileB, "B", LFS_O_CREAT | LFS_O_RDONLY) == 0);
}
```

If the compiled with optimizations and `fileA`/`fileB` are collapsed into the same stack location we would still pass the theoretical check for `is_valid`, but the second open would still result in a corrupted `lfs->mlist`, as you can see below.

```c
file->next = (lfs_file_t*)lfs->mlist;
lfs->mlist = (struct lfs_mlist*)file;
```

- `lfs->mlist` will point to `fileA`
- `file` will point to `fileB`
- `fileB->next` will point to `fileA`
- `lfs->mlist` will be updated to point to `fileB`

If `fileA` and `fileB` are the same location in memory, you can clearly see the list is circular and will result in an infinite recursion.

>3. Have operations that iterate over files assert is_valid on said files.

This doesn't really help identify the *cause*, and by the time you fail this assertion things have already gone pear shaped.  I would say it is only marginally more helpful than ending up stuck in an infinite loop and/or dereferencing invalid memory (which is the more likely real-world failure mode for this bug).
Also, it is entirely feasible in a case where you forget to close a file declared on the stack that the `is_valid` flag is **not** clobbered, but the `next` pointer **is** (or that the `is_valid` value is clobbered with a non-zero value and still passes the check).

>4. Have a counter num_open_files. Assert if it goes negative.

This would not help in the case of forgetting to *close* a file. (assuming you increment the counter on open and decrement on close).  It would catch the case of a double close, but I doubt that is a common bug.  As it stands, a double close in lfs would be well behaved except for a double free (which could be trivially fixed in any case) if you don't use `lfs_file_opencfg` and pass your own buffer.

>5. Have operations that iterate over files assert if they've looped "too many times" (num_open_files? Max number of files that theoretically can fit into the currently-open filesystem? Something like that.).

Just my opinion, but this introduces more problems than I believe it attempts to solve.
Suddenly we have put a cap on the number of files you can open.  This is one more thing the user needs to configure that previously was not even a concern (beyond the obvious performance problems if you decide to open thousands of files).
And honestly, beyond this very specific test case you have provided, I don't believe the most common failure mode for forgetting to close a file will be an infinite loop.  Chances are really good if you put the file handle on the stack and forget to close it, the failure mode is a hard fault caused by dereferencing invalid memory when chasing the `->next` pointer.

>Is this much better at catching this sort of thing that the current approach? Yes.

I respectfully disagree.

Having said that, there are definitely things that can be done in lfs to make this particular corner hurt less.  Unfortunately none of them come without some disadvantage.

# require memory allocation when opening files
- `lfs_file_t` basically becomes a handle to memory allocated by lfs internally
- `lfs_file_open` already uses `malloc` to allocate a cache buffer, so this could be implemented without any real overhead
- `lfs_file_opencfg` allows you to provide your own cache buffer and avoid allocation.  retaining this functionality would require some thought.

In this case, forgetting to close a file is simply a memory leak.  The system would continue to operate without issue, until you run out of memory.  The upshot is that if you only infrequently do file operations and your device resets semi regularly, you may never run into issues.  The downside is that it masks the problem *too well*.  If you have a bug, you may never find it.  Or even worse, it bites your users in the field.

# make `mlist` a doubly linked list

If each node points forward and backwards, you can pretty reliably detect memory corruption issues caused by forgetting to close a file allocated on the stack.

```c
for (lfs_file_t *f = (lfs_file_t*)lfs->mlist; f; f = f->next) {
}
```
[... 61 lines cut ...]

		return file(fs_, f);
	}

	// rest of relevant methods

private:
	lfs_t* fs_;
};
```

example usage
```c++
void test(filesystem& fs)
{
    auto f = fs.open("/test.txt", 0);
    if (!f) {
        printf("open failed, %i", f.error());
        return;
    }

    // do stuff
    f->read(...);
}  // f is automatically closed here
```

FreddieChopin, 2019-04-10:
> As I mentioned before, the most reliable way to avoid this kind of bug entirely is to use a c++ wrapper to the raw API (or just be very careful when using the existing api.

+1 here. Some things are not possible to solve reliably and completely in C, but in other languages it is much much easier. I've done it in similar way in [my project](https://github.com/DISTORTEC/distortos/blob/master/source/FileSystem/littlefs/LittlefsFile.cpp#L33), however I think I'll change it to be more "contract based" and use assertions instead of handling that automatically.

BTW - the proposed wrapper does double close when used properly (;

apmorton, 2019-04-10:
>BTW - the proposed wrapper does double close when used properly (;

balls...

For the record, my actual wrapper doesn't suffer this problem - I just broke it when dumbing it down to fit on the slide :/

My implementation uses `std::exchange` in the close routine on the `unique_ptr`, and all methods null check the unique_ptr before forwarding to lfs api's

meh, 2019-04-11:
Personally I'm fine with improving the documentation regarding this as the only fix, the technical drawbacks of trying to fix a breach of contract when it's the programmer misusing the library are way too big, and while "make everything explode" seems like a drastic failure mode everything would be broken anyway because the API contract has been breached.