littlefs-project/littlefs #152: Mounting corrupt filesystems and a couple of other questions.

tomvarghese, 2019-03-01:
Hello All,

I'm not an littlefs expert, but have done embedded stuff for a while, so not a complete newbie.  

My main issues is that we have seen a few instances where the flash gets corrupted, it gets mounted ok and then fails when trying to access a file that has disappeared or something.  Unfortunately, didn't think to save the flash at that time or investigate further.  Just reformatting fixed the problem.

The device does not threaten life and limb if the FS gets corrupted, we have the luxury of being able to just reformat and go.

All the same, my main questions are:

1) Is there a way to do a filesystem check before mounting?  Like fsck in Un*x systems or chkdsk on windows/dos that does a more thorough check before attempting to use the flash.
2) Alternately, Is there something I can do after mounting to verify file integrity to detect if a file has been corrupted in some way.

Secondary questions:
1) I've read there is some sort of wear leveling with littlefs, the main reason I switched from FAT. Can you point me at some docs?  Looked, but didn't see anything.  I know I can look at the code, but an overview and test results would be great.
2) We are using a run to completion kernel/multitasker, so when there are really long sequences, would b e nice to do something in wait loops.  Is there a way to register an idle task?  This could lead to a whole another can of worms, so not expecting much of an answer.

My system is an STM32F413 microcontroller that is connected to a Winbond W25Q32JV external SPI flash using the QuadSPI interface.  Works well, except that occasionally I've had writes fail (~2%) and can usually be fixed by just redoing the write without the need for an erase.  Reads have failed more rarely (~0.02%).  So writes have read to verify the write and all reads are done twice.

The reads are quite fast on the QuadSPI using four bits at a time. Writes are so slow anyway that it does not matter, using a single bit at a time. Both are working over DMA, so processor overhead is low, but usually have to wait anyway, so it really doesn't save much other than code to manually do the data movement.

FreddieChopin, 2019-03-01:
> Works well, except that occasionally I've had writes fail (~2%) and can usually be fixed by just redoing the write without the need for an erase. Reads have failed more rarely (~0.02%). So writes have read to verify the write and all reads are done twice.

I wouldn't say it "works well" if it fails 1 in 50 times (;

tomvarghese, 2019-03-01:
If I redo the SPI flash write without doing an erase, it completes the write and the data is good in the SPI flash. The retry happens automatically at a lower layer, so the FS does not see it.

FreddieChopin, 2019-03-01:
You definitely have some problem in your driver... Maybe automatic redo works around one manifestation of the issue, but who knows how many are still uncaught...

tomvarghese, 2019-03-05:
I agree with you that trying to gloss over a low level issue is not a good idea.  However, I can get a 100% reliable write and read at the lower levels by doing retries.  Why the retries are required, I don't know.  They are intermittent enough that getting a logic analyzer trace has not been successful. 

My guess would be the STM32 hardware or firmware. The layout is unlikely, the memory chip is literally right on top of the processor so the traces are very short, less than 1cm. Currently running at 50MHz, but I have slowed it down to 1MHz, same types of issues. It could be a problem with the chip, but have not seen any errata or other information that leads me to believe there is anything wrong with it. Does not mean that there couldn't be anything wrong either.  Sometimes hardware vendors are not very forthcoming with such information.  

If you have any code or test check the robustness of the SPI flash, that would be greatly appreciated.  Funny thing is that when I run my test code by itself (read/erase/write the entire 4MB flash multiple times), I see no issues whatsoever.  In my normal application there are a dozen peripherals that are active at the same time, so it could be some subtle bug in the hardware.

But all that aside.  Why LittleFs will mount a partition that is not really a partition or has errors is a completely different issue. I've seen the former happen when I pointed the software at partition that I was previously storing an array of a certain type of struct. Shouldn't have looked anything like a filesystem. This is really the reason I have asked if there was a more detailed consistency check function that I could run before mounting the filesystem.

Thanks for your time and efforts to help me with this problem.  I really appreciate the thoughtful insight you have provided.

tomvarghese, 2019-03-18:
Ok, I reworked my code to clean it up, found a couple of very small things that might have been producing a race condition.  Eliminated the write and erase errors.  But still getting read errors. So the retries are still in place.

tomvarghese, 2019-04-23:
Fixed it.  I switched over to using interrupts instead of DMA.  Works great for the full 100MHz data rate that the chip supports.

It is in the DMA section of the Errata sheet, 2.2.7.  No mention of it in the Quad-SPI section where I had looked originally.  The ST apps engineer pointed me at it and made some suggestions.  There is a workaround mentioned.  But I didn't look into it futher.  Too many other issues to deal with at the moment.

https://www.st.com/content/ccc/resource/technical/document/errata_sheet/0a/98/58/84/86/b6/47/a2/DM00037591.pdf/files/DM00037591.pdf/jcr:content/translations/en.DM00037591.pdf

I hope this helps somebody...

closed 2019-04-23 by tomvarghese

FreddieChopin, 2019-04-23:
> It is in the DMA section of the Errata sheet, 2.2.7.

There is no 2.2.7 in the linked document. Which problem mentioned in the errata did you have in mind? "2.1.10 - DMA2 data corruption when managing AHB and APB peripherals in aconcurrent way"?