littlefs-project/littlefs #153: Endless "bad block" loop when writing a file with size bigger than 512 bytes in one pass

desertkun, 2019-03-04:
I have this setup:
```c
	lfsconfig->read_size = 256;
	lfsconfig->prog_size = 256;
	lfsconfig->block_size = 4096;
	lfsconfig->block_count = ...
	lfsconfig->lookahead = 128;
```
So when I write file with size over 512 bytes in one call (assuming its twice the `prog_size`) I end up in endless loop:
```
littlefs has been mounted.
lfs debug:1414: Bad block at 5
lfs debug:1414: Bad block at 6
lfs debug:1414: Bad block at 7
lfs debug:1414: Bad block at 8
lfs debug:1414: Bad block at 9
lfs debug:1414: Bad block at 10
lfs debug:1414: Bad block at 11
lfs debug:1414: Bad block at 12
lfs debug:1414: Bad block at 13
.. and so on
```

Test example:
```c
#define TEST_SIZE (513)

...

uint8_t test[TEST_SIZE] = {[0 ... TEST_SIZE-1] = 0xAA};

  {
	lfs_file_t f;
	lfs_file_open(lfs, &f, "test", LFS_O_CREAT | LFS_O_RDWR);
	lfs_file_write(lfs, &f, test, TEST_SIZE);
	lfs_file_close(lfs, &f);
  }

  memset(test, 0, TEST_SIZE);

  {
	lfs_file_t f;
	lfs_file_open(lfs, &f, "test", LFS_O_RDONLY);
	lfs_file_read(lfs, &f, test, TEST_SIZE);
	lfs_file_close(lfs, &f);
  }

  for (uint32_t i = 0; i < TEST_SIZE; i++)
  {
	  if (test[i] != 0xAA)
	  {
		  printf("Whoops.\n");
		  return;
	  }
  }

  printf("Test OK\n");
```

If I define `TEST_SIZE` as less as 512, this works fine. 
If I split writes up in several write calls, it also works fine.

desertkun, 2019-03-05:
Okay turned out I had a second device on SPI line that interfered with the communication.

desertkun, 2019-03-06:
OK its just me being not able to correctly read documencation, you're not supposed to do that, you have to write and do `lfs_file_sync`.

closed 2019-03-06 by desertkun

FreddieChopin, 2019-03-06:
> you're not supposed to do that, you have to write and do lfs_file_sync.

?

You definitely can do writes bigger than read/program/erase block size. Doing manual `lfs_file_sync` before `lfs_file_close` is also not required.