littlefs-project/littlefs #155: Shifting signed 32-bit value by 31 bits is undefined behaviour

keck-in-space, 2019-03-12:
Hello,

I was using CppCheck recently and came across this issue:

```c
// reset the next bit if we need to
ptag ^= (lfs_tag_chunk(tag) & 1) << 31;
```

CppCheck calls this an error with the explanation: Shifting signed 32-bit value by 31 bits is undefined behaviour

okchip73, 2019-03-13:
Hello,
I think the IAR compiler seems to interpret it differently.

I have problem every restart my system.
lfs error:976: Corrupted dir pair at 0 1

My problem is cleared by change below
ptag ^= ((lfs_tag_chunk(tag) & 1) << 31);

Thanks a lot.

C47D, 2019-03-13:
AFAIK shift operations are not recommended to be used with signed types (see [MISRA Rule 12.7](http://caxapa.ru/thumbs/468328/misra-c-2004.pdf)).

armandas, 2019-03-15:
+1 Was going to report this too.

geky, 2019-04-01:
This is quite a subtle bug!

I was confused because `lfs_tag_chunk` is defined to return `uint8_t`. So I thought the value would be unsigned.
https://github.com/ARMmbed/littlefs/blob/bdff4bc59eb69bc403e0f5878e72a96a0a21a190/lfs.c#L299

Normally, `unsigned & signed` would keep the result as an unsigned number, but it turns out this _isn't_ the case if the signed number is a larger type. So `uint8_t & signed` actually gets converted into a signed number.

https://stackoverflow.com/a/17833338

The fix is to use `1U` there, I'll push up a fix when I can. Thanks for raising the issue.
``` c
// reset the next bit if we need to
ptag ^= (lfs_tag_chunk(tag) & 1U) << 31;
```

geky, 2019-04-09:
Pushed up a fix on the v2-alpha branch (https://github.com/ARMmbed/littlefs/issues/155).

Out of curiosity, @keck-in-space, @armandas, what flags were you guys using for cppcheck? It looks interesting but I wasn't able to get the same warnings. It may be valuable to add to CI.

armandas, 2019-04-10:
Nothing special here:
`cppcheck --enable=all --inconclusive --xml --xml-version=2`

keck-in-space, 2019-04-10:
Thanks, @geky. I'm using the default config on the CppCheck GUI available here: http://cppcheck.net.